Stefal/rtkbase #14: ubxconfig.sh sometimes doesn't find the version until the last try

Stefal, 2019-11-22:
```
root@centipede:/rtkbase# ./ubxconfig.sh /dev/ttyACM0 receiver_cfg/U-Blox_ZED-F9P_rtkbase.txt
stty: /dev/ttyACM0: unable to perform all requested operations
dos2unix: converting file receiver_cfg/U-Blox_ZED-F9P_rtkbase.txt to Unix format...

Checking compatibility...
Version mismatch! Checking again...
Version mismatch! Checking again...
Version mismatch! Checking again...
Version mismatch! Checking again...
Version mismatch! Checking again...
Version mismatch! Checking again...
Version mismatch! Checking again...
Version mismatch! Checking again...
xxd: Broken pipe
cat: write error: Broken pipe
Version match!
Configuring u-blox GPS at /dev/ttyACM0 from receiver_cfg/U-Blox_ZED-F9P_rtkbase.txt...
```

Stefal, 2019-11-22:
It seems related to the `export stty` line:
https://github.com/Stefal/rtkbase/blob/66dd2b757b1b634d25558651360a698bd0bf25a2/receiver_cfg/ubxconfig.sh#L33

Stefal, 2020-02-24:
It doesn't work at all with armbian buster

Stefal, 2020-04-22:
It works, but don't forget to stop the str2str_tcp service before using `ubxconfig.sh`
If it doesn't work, please send me the output of:
`stty -F /dev/your_gps -g`

Stefal, 2020-04-22:
On my Orange Pi Zero with a F9P connected via Usb, and without any str2str communication started since boot, I get:
```
basegnss@orangepizero:~$ stty -F /dev/ttyACM0
speed 9600 baud; line = 0;
-brkint -imaxbel
```
After a few minutes with str2str running then stopping it:
```
basegnss@orangepizero:~/rtkbase$ stty -F /dev/ttyACM0
speed 115200 baud; line = 0;
min = 0; time = 0;
-brkint -icrnl -imaxbel
-opost -onlcr
-isig -icanon -iexten -echo -echoe -echok -echoctl -echoke
```

jancelin, 2020-04-27:
Working well if str2str don't run!

```
centipede@centipede in /rtkbase
$ sudo ./ubxconfig.sh /dev/ttyACM0 receiver_cfg/U-Blox_ZED-F9P_rtkbase.txt --force
dos2unix: converting file receiver_cfg/U-Blox_ZED-F9P_rtkbase.txt to Unix format...
Checking compatibility...
Firmware mismatch but product match!
Product is ZED-F9P
Trying to send the settings...
Configuring u-blox GPS at /dev/ttyACM0 from receiver_cfg/U-Blox_ZED-F9P_rtkbase.txt...
Resetting to factory defaults...
Sending UBX message: UBX-CFG-CFG-REVERTDEFAULT B5 62 06 09 0D 00 FF FF 00 00 00 00 00 00 FF FF 00 00 17 2F AE
ACKNOWLEDGED
Restoring configuration from file...
Sending UBX message: CFG-MSG B5 62 06 01 08 00 21 08 00 00 00 00 00 00 38 E3
ACKNOWLEDGED
Sending UBX message: CFG-MSG B5 62 06 01 08 00 0A 0B 00 00 00 00 00 00 24 40
ACKNOWLEDGED
Sending UBX message: CFG-MSG B5 62 06 01 08 00 0A 37 00 00 00 00 00 00 50 74
...
```

jancelin, 2020-04-27:
#25

Stefal, 2020-04-27:
I don't know if flashing the F9P is the best solution. See #19

Stefal, 2020-05-06:
I still have some problems with ubxconfig.sh. It doesn't detect the F9P 90% of time. The problem seems in this line:
https://github.com/Stefal/rtkbase/blob/f9c157f7f58d86b01dd0eb86c231cbd7e35a7498/ubxconfig.sh#L70
the grep command doesn't receive any end of line and then doesn't output any value.
I'm switching to ubxtool from gpsd: https://github.com/Stefal/rtkbase/commit/887b342239d5d565a0e62583303cfc7c7c078a57

I close this issue.

closed 2020-05-06 by Stefal