Stefal/rtkbase #19: On the fly gnss receiver configuration? Or write settings in flash?

Stefal, 2020-04-23:
We have to way to configure the U-blox gnss receivers:
- Send the settings to flash with `ubxconfig.sh`. It's slow, it doesn't work everytime, but you do have to run it only once.
- Send the settings to ram during each startup with the `str2str -c mySettings.cmd`. It's easier if you already have the good settings file, but it needs to use the rtklibexplorer's RTKLIB [fork](https://github.com/rtklibexplorer/RTKLIB/)

jancelin, 2020-04-27:
That's a good question. 
I like the idea very much because it may be possible to change parameters at each launch quite easily.
It forces you to switch to rtklibexplorer at 100%. I'll run a stability test. I had noticed some instabilities a few months ago compared to version 2.3.4.

Stefal, 2020-05-01:
There is another option:
- Modifying run_cast.sh to send the settings to the receiver before we start `str2str -in serial:// ...`

This way, we can stay on the official rtklib repo and still have a way to change the receiver settings

Stefal, 2020-05-06:
Ok, I will stay with sending the settings in flash memory, but with ubxtool (from gpsd) instead of ubxconfig. -> https://github.com/Stefal/rtkbase/commit/887b342239d5d565a0e62583303cfc7c7c078a57
I will add a feature to send some settings on the fly later.

Stefal, 2020-06-05:
It's possible to send some settings with `ubxtool` even thought str2str is running. `Ubxtool` doesn't receive the 'ACK' but it works. It could be useful in special cases.
Closing.

closed 2020-06-05 by Stefal