Stefal/rtkbase #20: The flask web server needs root

Stefal, 2020-04-23:
It needs to be root to start/stop some systemd's services. We need to find a way to run it without root.

kikislater, 2020-05-29:
I think link below cover the need : 
Install pip as user, custom paths for requirements, launch pip command as user => 

https://medium.com/@dorukgezici/how-to-setup-python-flask-app-on-shared-hosting-without-root-access-e40f95ccc819

Stefal, 2020-05-29:
The issue is not about the way to run Flask as a standard user (it's already possible) but how to start/stop systemd services, reboot/shutdown the sbc without starting flask as 'root'.
The only solution I have in mind is edit the sudoer file to allow `systemctl` without password. But I'm not sure it's a good idea.

Perhaps it's possible with a d-bus [configuration file](https://dbus.freedesktop.org/doc/dbus-daemon.1.html), but it's out of my skill so far.

kikislater, 2020-05-29:
You could run `systemctl` as user : 
`systemctl --user [start|stop|enable|disable|status] name.service`
service files should be placed at : 
`~/.config/systemd/user/name.service`

https://computingforgeeks.com/how-to-run-systemd-service-without-root-sudo/

Stefal, 2020-05-29:
Already tested, but it's a no go: user services can't depend from system-wide services

Stefal, 2023-12-12:
Since Debian Bookworm it's possible to add some policykit rules to enable a group to manage some services.

 - install polkit `sudo apt install policykit-1`
Let's say our user is in the group "rtkbase", create a file named 99-rtkbase.rules in /etc/polkit-1/rules.d/ with this content: 
```
polkit.addRule(function(action, subject) {
    if (action.id == "org.freedesktop.systemd1.manage-units" &&
       action.lookup("unit") == "str2str_tcp.service" &&
       subject.isInGroup("rtkbase"))
       {
         return polkit.Result.YES;
       }
    });
```
Now you can start/stop/... the str2str_tcp service.
(using subject.user("rtkbase") doesn't work for stopping services)

Another way to add this rule to all str2str_* services: 
```
polkit.addRule(function(action, subject) {
    if (action.id == "org.freedesktop.systemd1.manage-units" &&
       RegExp('str2str_[A-Za-z0-9]+.service').test(action.lookup("unit")) === true &&
       subject.isInGroup("rtkbase"))
       {
         return polkit.Result.YES;
       }
    });
```