Stefal/rtkbase #48: raw logfile archive: tar or zip?

Stefal, 2020-05-26:
+ tar: more common on linux platforms
+ zip: more common on windows platforms

kikislater, 2020-05-29:
If both are considered, most of gnu/linux distribution now support zstd which speed up in making archive. 
tar archive have a large impact on cpu load and could decrease performance. Example in one distribution switching from tar.xz to zstd : 1300% speedup => https://www.archlinux.org/news/now-using-zstandard-instead-of-xz-for-package-compression/

Stefal, 2020-05-29:
Hi @kikislater! and thanks for your comment.
It seems that the speedup is only for decrompression time, right?

As the tar script runs only once a day, I don't think that compression time or cpu usage is a deal breaker. It's more about "Which platform is used by the user whom will download raw archives" and "which archive format" is the easiest to extract for these users.

If I had to choose right now, I think I will switch to the 'zip' format, but the question remains opened.

kikislater, 2020-05-29:
> If I had to choose right now, I think I will switch to the 'zip' format, but the question remains opened.

Good !

No, on creation as well, I notice several speed up in several configurations : i7 4790k or i7 2670qm or i7 8550u

Stefal, 2020-06-05:
End users seem to prefer .zip. Let's switch: 
https://github.com/Stefal/rtkbase/commit/2d795caba0636211c7606a46349872e937e94105

closed 2020-06-05 by Stefal