Stefal/rtkbase #49: FR: Get datetime from receiver

Stefal, 2020-05-28:
problem: When the base station isn't connected to the internet, the internal clock is unsync, then str2str_file doesn't start.
solution: get date and time from the gnss receiver. There are many methods to do this. One of these is to modify https://github.com/Stefal/rtkbase/blob/web_gui/web_app/reach_tools/gps_time.py
- don't use serial port, but tcp
- parse msg_nav_timeutc or msg_nav_pvt

It will works only with u-blox receiver.

Other options: gpsd + chrony

Stefal, 2020-05-28:
The actual gpsd release (v3.17) we can get with apt, is too old to works with a F9P.

kikislater, 2020-05-29:
3.20 is availaible in repo : http://archive.raspbian.org/raspbian/pool/main/g/gpsd/

closed 2020-06-02 by Stefal