Stefal/rtkbase #59: Too much log from str2str

Stefal, 2020-06-05:
All str2str instances write their output in syslog. That's too much unuseful data like this:
```
Jun 05 16:32:09 orangepizero run_cast.sh[583]: 2020/06/05 16:32:27 [CW---]   31076196 B   31042 bps (0) 127.0.0.1 (1) waiting...
Jun 05 16:32:09 orangepizero run_cast.sh[598]: 2020/06/05 16:32:27 [CC---]   31076196 B   31089 bps (0) /dev/ttyACM0 (1) 3 clients
Jun 05 16:32:09 orangepizero run_cast.sh[615]: 2020/06/05 16:32:27 [CC---]   31049708 B   30883 bps (0) 127.0.0.1
Jun 05 16:32:14 orangepizero run_cast.sh[583]: 2020/06/05 16:32:32 [CW---]   31095024 B   30660 bps (0) 127.0.0.1 (1) waiting...
Jun 05 16:32:14 orangepizero run_cast.sh[598]: 2020/06/05 16:32:32 [CC---]   31095024 B   30818 bps (0) /dev/ttyACM0 (1) 3 clients
Jun 05 16:32:14 orangepizero run_cast.sh[615]: 2020/06/05 16:32:32 [CC---]   31068984 B   30404 bps (0) 127.0.0.1
Jun 05 16:32:19 orangepizero run_cast.sh[583]: 2020/06/05 16:32:37 [CW---]   31113744 B   31075 bps (0) 127.0.0.1 (1) waiting...
Jun 05 16:32:19 orangepizero run_cast.sh[598]: 2020/06/05 16:32:37 [CC---]   31113744 B   35438 bps (0) /dev/ttyACM0 (1) 3 clients
Jun 05 16:32:19 orangepizero run_cast.sh[615]: 2020/06/05 16:32:37 [CC---]   31087256 B   31091 bps (0) 127.0.0.1
Jun 05 16:32:24 orangepizero run_cast.sh[583]: 2020/06/05 16:32:42 [CW---]   31133380 B   29552 bps (0) 127.0.0.1 (1) waiting...
Jun 05 16:32:24 orangepizero run_cast.sh[598]: 2020/06/05 16:32:42 [CC---]   31133380 B   29370 bps (0) /dev/ttyACM0 (1) 3 clients
Jun 05 16:32:24 orangepizero run_cast.sh[615]: 2020/06/05 16:32:42 [CC---]   31106892 B   29163 bps (0) 127.0.0.1
Jun 05 16:32:29 orangepizero run_cast.sh[583]: 2020/06/05 16:32:47 [CW---]   31151772 B   28972 bps (0) 127.0.0.1 (1) waiting...
Jun 05 16:32:29 orangepizero run_cast.sh[598]: 2020/06/05 16:32:47 [CC---]   31151772 B   29463 bps (0) /dev/ttyACM0 (1) 3 clients
Jun 05 16:32:29 orangepizero run_cast.sh[615]: 2020/06/05 16:32:47 [CC---]   31125284 B   29227 bps (0) 127.0.0.1
Jun 05 16:32:34 orangepizero run_cast.sh[583]: 2020/06/05 16:32:52 [CW---]   31171532 B   31473 bps (0) 127.0.0.1 (1) waiting...
Jun 05 16:32:34 orangepizero run_cast.sh[598]: 2020/06/05 16:32:52 [CC---]   31171532 B   31856 bps (0) /dev/ttyACM0 (1) 3 clients
Jun 05 16:32:34 orangepizero run_cast.sh[615]: 2020/06/05 16:32:52 [CC---]   31145044 B   31570 bps (0) 127.0.0.1
Jun 05 16:32:39 orangepizero run_cast.sh[583]: 2020/06/05 16:32:57 [CW---]   31190444 B   30850 bps (0) 127.0.0.1 (1) waiting...
Jun 05 16:32:39 orangepizero run_cast.sh[598]: 2020/06/05 16:32:57 [CC---]   31190444 B   31088 bps (0) /dev/ttyACM0 (1) 3 clients
Jun 05 16:32:39 orangepizero run_cast.sh[615]: 2020/06/05 16:32:57 [CC---]   31163956 B   31104 bps (0) 127.0.0.1
Jun 05 16:32:44 orangepizero run_cast.sh[583]: 2020/06/05 16:33:02 [CW---]   31209492 B   30852 bps (0) 127.0.0.1 (1) waiting...
Jun 05 16:32:44 orangepizero run_cast.sh[598]: 2020/06/05 16:33:02 [CC---]   31209492 B   30725 bps (0) /dev/ttyACM0 (1) 3 clients
Jun 05 16:32:44 orangepizero run_cast.sh[615]: 2020/06/05 16:33:02 [CC---]   31183508 B   30771 bps (0) 127.0.0.1
Jun 05 16:32:49 orangepizero run_cast.sh[583]: 2020/06/05 16:33:07 [CW---]   31228276 B   37328 bps (0) 127.0.0.1 (1) waiting...
Jun 05 16:32:49 orangepizero run_cast.sh[598]: 2020/06/05 16:33:07 [CC---]   31228276 B   30594 bps (0) /dev/ttyACM0 (1) 3 clients
Jun 05 16:32:49 orangepizero run_cast.sh[615]: 2020/06/05 16:33:07 [CC---]   31201788 B   30624 bps (0) 127.0.0.1
```

Stefal, 2020-06-05:
https://www.ctrl.blog/entry/systemd-log-levels.html

Stefal, 2020-06-06:
fixed

closed 2020-06-06 by Stefal

satel-kalletuulos, 2023-09-12:
This ticket has already been closed, but I'm commenting still: even with limiting, there is too much logging to syslog. With current options, it is not possible to completely disable the regular output from str2str.

commit c5f57723 (2020-06-06): Limiting log messages
Fix #59