Stefal/rtkbase #70: add a buton for detect and configure gnss

jancelin, 2020-06-17:
somme people upgrade F9P and are forced to switch to ssh to put the parameters back into the receiver.
Solution: add a web button with a function 
 ```sudo /home/basegnss/rtkbase/tools/install.sh --detect-usb-gnss --configure-gnss```

Stefal, 2020-06-17:
Yes, I can do that.
But users could save the F9P configuration before flashing it with a new firmware.

cofhal, 2021-01-17:
How can I use an MT8 receiver and what is the basegnss login and password?

Stefal, 2022-03-13:
This feature is a work in progress in the branch https://github.com/Stefal/rtkbase/tree/detect_button
Sadly it doesn't work when server.py is launched with a systemd service because https://github.com/Stefal/rtkbase/blob/2e939c89f12d5bac538716947c29b4d74bcbb0a7/tools/install.sh#L397 fails when it is called from https://github.com/Stefal/rtkbase/blob/2e939c89f12d5bac538716947c29b4d74bcbb0a7/web_app/server.py#L529

let's use this script test_user.sh
```
#!/bin/bash

echo 'logname  : ' $(logname)
echo  'whoami  : ' $(whoami)
echo '$LOGNAME  : ' $LOGNAME
```
`./test_user.sh`
```
logname  :  pi
whoami  :  pi
$LOGNAME  :  pi
```
`sudo test_user.sh`
```
logname  :  pi
whoami  :  root
$LOGNAME  :  root
```
from a systemd service, with User=root
```
Mar 13 13:25:04 raspberrypi test_user.sh[10085]: logname: no login name
Mar 13 13:25:04 raspberrypi test_user.sh[10084]: logname:
Mar 13 13:25:04 raspberrypi test_user.sh[10084]: whoami:  root
Mar 13 13:25:04 raspberrypi test_user.sh[10084]: $LOGNAME:  root
```
I see 3 ways to fix this:
1. in `install.sh` if the `logname` test fail, test with `whoami` too
2. Add a `--user` argument to `install.sh`
3. `install.sh` could get the user value from the `settings.conf`

closed 2023-02-20 by Stefal