Stefal/rtkbase #71: add a pin with Base coordinates  on map

jancelin, 2020-06-17:
add a pin with Base coordinates  on map

Stefal, 2020-06-17:
I have this feature in mind since a few weeks. I search a good icon to the real location, something like a target?

Stefal, 2020-06-17:
Something like this?

[image]
Perhaps a larger crosshair?

Stefal, 2020-06-19:
Feature added in https://github.com/Stefal/rtkbase/commit/fff08bbc158d063489a00a4afe2b5b0b8b3f6362

jancelin, 2020-06-19:
[image]
it's possible to put basertk icon on top of the other?

Stefal, 2020-06-19:
Yes, it's possible, but the rendering is....strange:
[image]

Stefal, 2020-06-19:
With a bigger crosshair (24x24 px)?
[image]

Stefal, 2020-06-23:
Crosshair extended to 24x24: https://github.com/Stefal/rtkbase/commit/d017e366e1ce5e06e4bfd357a4a58a3186c4627f

closed 2020-06-23 by Stefal