Stefal/rtkbase #85: Installer failing to clone RTKLIB

domtisdell, 2020-07-11:
Probably my issue but having problems getting installer to clone RTKLIB, both as part of --all option and separately installing as Step 2.  

Removing the last subfolder from the path does the trick for Step 2 i.e. $ git clone -b rtklib_2.4.3 https://github.com/tomojitakasu/RTKLIB

Stefal, 2020-07-12:
Hi!
Do you remember the error message?
Did you use `sudo` to run `install.sh`?

domtisdell, 2020-07-12:
No sorry. I removed the **BOLD** part of the instructions below when installing manual to get it to work.
2. Install RTKLIB with sudo ./install.sh --rtklib, or:
clone RTKlib
$ cd ~
$ git clone -b rtklib_2.4.3 https://github.com/tomojitakasu/RTKLIB**/rtklib_2.4.3**

For Step 2 and 3 I had problems using the sudo ./install.sh to connect to GitHub. Removing the login/user section prior to the clone / wget from the install.sh allowed both Steps 2 and 3 to run.

Stefal, 2020-07-12:
I just fixed an error in the rtklib manual installation instructions: https://github.com/Stefal/rtkbase/commit/32710baa27cab79801827b033cee48500baa65bb

Thank you for noticed me that problem.

About the step 2 and 3, one possible explanation is that the `logname` command didn't return anything. It could happen on a fresh Os installation if you didn't reboot after the standard user creation.

Stefal, 2020-07-12:
If it's ok for you, you can close this issue.

domtisdell, 2020-07-12:
Thanks Stefal.

closed 2020-07-12 by domtisdell