Stefal/rtkbase #90: Add average position to fix base coordinates

PPiquemal, 2020-07-21:
Sometime we want to setup a base station on an unknown location. The possibility to observe and average coordinates during a short time from the setup interface allows the immediate use with limited absolute accuracy, but centimetric relative  accuracy, before further PPP calculations.

Stefal, 2020-07-21:
Hi!
The realtime compute mode used to locate the base is PPP Static. See https://github.com/Stefal/rtkbase/blob/master/web_app/rtklib_configs/rtkbase_single_default.conf
The more you wait, the more the location is accurate, so I don't see any improvement to calculate an average value.
Do you have a better solution?

PPiquemal, 2020-07-21:
Hi Stefal,

Thank you ! So I understand that there is no coordinates to enter in interface on unknown point or first installation, as rtklib alone is refining the coordinates. So there is no improvement in averaging. Coordinates should be entered only after a post processing (PPP or baseline/known point).

Stefal, 2020-07-22:
If you don't do any realtime job, yes, you can just log the data.
But if a rover need a valid realtime rtcm stream, you should enter some coordinates in the settings page. It doesn't matter if it's not really accurate, but is the coordinate are completely wrong, I'm afraid the rover can't reach a rtk 'fix".

closed 2021-01-12 by Stefal