Stefal/rtkbase #91: Problem about pystemd ?

Tanggeshi, 2020-07-22:
I am installing rtkbase-master. But I meet this problem: when I running "sudo ./install --all", it works for several step, but at last, it shows following information and then stop install and exit. 

ERROR: pystemd-0.8.1590398158-cp37-cp37m-linux_armv7l.whl is not a supported wheel on this platform

How to solve this problem? Thank you!

Stefal, 2020-07-22:
Hi!
Do you use a x86 platform? If yes, see #81.

Tanggeshi, 2020-07-22:
Thank you for reply!
My system is:  Linux orangepi4 4.4.179-g7b2dd521-dirty #5 SMP Tue Mar 24 14:07:36 CST 2020 aarch64 aarch64 aarch64 GNU/Linux.  
Also, I had finished it following #81, but there are any other problem. I copy all information as follow:

################################
orangepi@orangepi4:~$ sudo -H ./rtkbase-master/tools/install.sh --start-services
Installation options:  --start-services
################################
STARTING SERVICES
################################
Failed to start rtkbase_web.service: Unit rtkbase_web.service is not loaded properly: Exec format error.
See system logs and 'systemctl status rtkbase_web.service' for details.
A dependency job for gpsd.service failed. See 'journalctl -xe' for details.
################################
END OF INSTALLATION
You can open your browser to http://10.42.1.1 10.42.0.1
################################
orangepi@orangepi4:~$ systemctl status rtkbase_web.service
● rtkbase_web.service - RTKBase Web Server
   Loaded: error (Reason: Exec format error)
   Active: inactive (dead)

Jul 22 12:41:35 orangepi4 systemd[1]: /etc/systemd/system/rtkbase_web.service:8: Executable path is not absolute: python3 /home/orangepi/rtkbase/web_app/server.py
Jul 22 12:41:35 orangepi4 systemd[1]: /etc/systemd/system/rtkbase_web.service:8: Executable path is not absolute: python3 /home/orangepi/rtkbase/web_app/server.py
Jul 22 12:41:55 orangepi4 systemd[1]: /etc/systemd/system/rtkbase_web.service:8: Executable path is not absolute: python3 /home/orangepi/rtkbase/web_app/server.py
Jul 22 12:41:55 orangepi4 systemd[1]: /etc/systemd/system/rtkbase_web.service:8: Executable path is not absolute: python3 /home/orangepi/rtkbase/web_app/server.py
Jul 22 12:42:14 orangepi4 systemd[1]: /etc/systemd/system/rtkbase_web.service:8: Executable path is not absolute: python3 /home/orangepi/rtkbase/web_app/server.py
Jul 22 12:42:14 orangepi4 systemd[1]: /etc/systemd/system/rtkbase_web.service:8: Executable path is not absolute: python3 /home/orangepi/rtkbase/web_app/server.py
Jul 22 12:42:14 orangepi4 systemd[1]: /etc/systemd/system/rtkbase_web.service:8: Executable path is not absolute: python3 /home/orangepi/rtkbase/web_app/server.py
Jul 22 12:42:15 orangepi4 systemd[1]: /etc/systemd/system/rtkbase_web.service:8: Executable path is not absolute: python3 /home/orangepi/rtkbase/web_app/server.py
Jul 22 12:42:15 orangepi4 systemd[1]: /etc/systemd/system/rtkbase_web.service:8: Executable path is not absolute: python3 /home/orangepi/rtkbase/web_app/server.py
Jul 22 12:44:08 orangepi4 systemd[1]: /etc/systemd/system/rtkbase_web.service:8: Executable path is not absolute: python3 /home/orangepi/rtkbase/web_app/server.py
orangepi@orangepi4:~$ ls /home/orangepi/rtkbase/web_app/server.py
/home/orangepi/rtkbase/web_app/server.py
orangepi@orangepi4:~$ journalctl -xe
Jul 22 12:46:01 orangepi4 ModemManager[4345]: <info>  Couldn't check support for device '/sys/devices/platform/fe3a0000.usb/usb8/8-1': not supported by any plugin
Jul 22 12:46:09 orangepi4 systemd[1]: str2str_tcp.service: Service hold-off time over, scheduling restart.
Jul 22 12:46:09 orangepi4 systemd[1]: str2str_tcp.service: Scheduled restart job, restart counter is at 4.
-- Subject: Automatic restarting of a unit has been scheduled
-- Defined-By: systemd
-- Support: http://www.ubuntu.com/support
-- 
-- Automatic restarting of the unit str2str_tcp.service has been scheduled, as the result for
-- the configured Restart= setting for the unit.
Jul 22 12:46:09 orangepi4 systemd[1]: Stopped RTKBase Tcp.
-- Subject: Unit str2str_tcp.service has finished shutting down
-- Defined-By: systemd
-- Support: http://www.ubuntu.com/support
-- 
-- Unit str2str_tcp.service has finished shutting down.
Jul 22 12:46:09 orangepi4 systemd[1]: Starting RTKBase Tcp...
-- Subject: Unit str2str_tcp.service has begun start-up
-- Defined-By: systemd
-- Support: http://www.ubuntu.com/support
-- 
-- Unit str2str_tcp.service has begun starting up.
Jul 22 12:46:09 orangepi4 run_cast.sh[4356]: mkdir: missing operand
Jul 22 12:46:09 orangepi4 run_cast.sh[4356]: Try 'mkdir --help' for more information.
Jul 22 12:46:09 orangepi4 run_cast.sh[4356]: /home/orangepi/rtkbase/run_cast.sh: line 25: -in: command not found
Jul 22 12:46:09 orangepi4 systemd[1]: Started RTKBase Tcp.
-- Subject: Unit str2str_tcp.service has finished start-up
-- Defined-By: systemd
-- Support: http://www.ubuntu.com/support
-- 
-- Unit str2str_tcp.service has finished starting up.
-- 
-- The start-up result is RESULT.
Jul 22 12:46:09 orangepi4 systemd[1]: str2str_tcp.service: Main process exited, code=exited, status=127/n/a
Jul 22 12:46:09 orangepi4 systemd[1]: str2str_tcp.service: Failed with result 'exit-code'.
Jul 22 12:46:09 orangepi4 dhclient[1142]: DHCPDISCOVER on eth0 to 255.255.255.255 port 67 interval 3 (xid=0x5e47bf3e)
Jul 22 12:46:12 orangepi4 dhclient[1142]: DHCPDISCOVER on eth0 to 255.255.255.255 port 67 interval 6 (xid=0x5e47bf3e)
Jul 22 12:46:18 orangepi4 dhclient[1142]: DHCPDISCOVER on eth0 to 255.255.255.255 port 67 interval 6 (xid=0x5e47bf3e)

##############################
Would you please give me some advices to solve these problem! Thank you!

Stefal, 2020-07-22:
Your system is a 64bits arm. I didn't test.
But as there is no aarch64 release for pystemd v0.8, it won't work.

The error about `Executable path is not absolute` is another thing I should look at.

hcwinsemius, 2020-08-27:
Dear @Stefal 

Software looks great for a lot of purposes, so want to give it a try with ardusimple ZED-F9P kits. I have the same problem with rapberry pi OS on a raspi zero. It won't get past the mmessage:
```
ERROR: pystemd-0.8.1590398158-cp37-cp37m-linux_armv7l.whl is not a supported wheel on this platform
```
any chance you know a solution? I am definitely on a 32-bit platform.

Cheers

Stefal, 2020-08-27:
Hi @hcwinsemius 

The raspberry pi zero is an armv6, and the wheel I've builded is for armv7.

As long as the pystemd dev don't publish a new release, you have to build pystemd manually.
I'm sorry for that.

hcwinsemius, 2020-08-27:
ah, ok that might explain, I'll try on a raspi3b+ then. Any chance there is a wheel for raspi zero?

Stefal, 2020-08-27:
If a new pystemd release is soon to be publish, It won't be needed to build a wheel, but you can create one yourself.
I don't recommend wifi network for a base station. Lan + POE is my preferred method, and if you want a cheap SBC, take a look at the orange pi zero (512mb).

Stefal, 2020-09-05:
I've built and added the pystemd wheels for armv6l (pi zero) and aarch64 platform: https://github.com/Stefal/rtkbase/tree/dev/tools
If you to test, you can install a wheel with `sudo python3 -m pip path_to_the_wheel`
The next release will check the platform and install the correct wheel:
https://github.com/Stefal/rtkbase/blob/7a45d0e98008d7e25c14b8b9d032151373102704/tools/install.sh#L241:250
```
platform=$(uname -m)
      if [[ $platform =~ 'armv7' ]]
      then
        python3 -m pip install ${rtkbase_path}/tools/pystemd-0.8.1590398158-cp37-cp37m-linux_armv7l.whl
      elif [[ $platform =~ 'aarch64' ]]
      then
        python3 -m pip install ${rtkbase_path}/tools/pystemd-0.8.1599238382-cp37-cp37m-linux_aarch64.whl
      elif [[ $platform =~ 'armv6' ]]
      then
        python3 -m pip install ${rtkbase_path}/tools/pystemd-0.8.1599232456-cp37-cp37m-linux_armv6l.whl
      fi
```

rlesac, 2020-11-24:
Banana Pi running Ubuntu Bionic with Armbian Linux 4.19.62-sunxi.
I also get:
```
sudo -H python3 -m pip install pystemd-0.8.1590398158-cp37-cp37m-linux_armv7l.whl
ERROR: pystemd-0.8.1590398158-cp37-cp37m-linux_armv7l.whl is not a supported wheel on this platform.
```
```
bananapi:~/rtkbase/rtkbase/tools$ cat /proc/cpuinfo
processor       : 0
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 27.78
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xc07
CPU revision    : 4

processor       : 1
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 27.78
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xc07
CPU revision    : 4

Hardware        : Allwinner sun7i (A20) Family
Revision        : 0000
Serial          : 1651664b09419b49

bananapi:~/rtkbase/rtkbase/tools$ uname -m
armv7l
```

Stefal, 2020-11-28:
Hi @rlesac,
Since the latest rtkbase release, the pystemd dev release a new version. Could you install it from pypi ?

Stefal, 2021-01-13:
The latest release use the official pystemd 0.8.0 release.
https://github.com/Stefal/rtkbase/releases/tag/2.2.0

closed 2021-01-13 by Stefal