Stefal/rtkbase #100: add skyplot to web view

iceberg1369, 2020-08-21:
it would be nice to include a skyplot in status page

Stefal, 2020-08-21:
Do you know a nice already existing javascript library for this feature?

RichStokesNADA, 2021-05-06:
Hi Stefal

May I start by saying this is fantastic! It would be great if you could look into the amount of work required to add the skyplot that iceberg1369 mentions above. The graph in the bottom right of http://www.taroz.net/GNSS-Radar.html (link from the git repo above) would be particularly useful!
Please can you advise if this is likely to be added in future revisions or does it seem unlikely? 
Thanks a lot

[image]