Stefal/rtkbase #101: missing dep in install.sh

hcwinsemius, 2020-08-28:
Using raspberry OS lite, one dependency seems missing in install_dependencies() preventing the python requirements to to be installed. Probably this lib is default available in more rich OS images.

If you add sudo apt install libffi-dev

this is fixed.

Stefal, 2020-08-29:
Hi!
On my running RTKBase, `libffi-dev` isn't installed.
Which python module need it? And what is your Os?

hcwinsemius, 2020-08-31:
Hmm I should've mentioned that. I am pretty sure it is a dependency of cryptography. I looked up the deps list.

https://gist.github.com/DavidEGrayson/007688ce94220048b07c

I use raspberry OS lite (so a headless Pi) on a 3b+ pi.

Stefal, 2020-08-31:
What it this raspberry OS lite? Could you provide a link?

Stefal, 2021-01-13:
It should work with the latest release. https://github.com/Stefal/rtkbase/releases/tag/2.2.0

closed 2021-01-13 by Stefal