Stefal/rtkbase #102: RINEX file v2.11 for PPP generation fails

hcwinsemius, 2020-08-28:
This sounds like a near-ideal functionality for easy PPP. But unfortunately there seems to be a bug in it.

I get the following HTML formatted error on screen:

**Failed!**

End-of-central-directory signature not found. Either this file is not a zipfile, or it constitutes one disk of a multi-part archive. In the latter case the central directory and zipfile comment will be found on the last disk(s) of this archive. unzip: cannot find zipfile directory in one of 2020-08-28_17-26-14_GNSS-1.ubx.tag or 2020-08-28_17-26-14_GNSS-1.ubx.tag.zip, and cannot find 2020-08-28_17-26-14_GNSS-1.ubx.tag.ZIP, period. End-of-central-directory signature not found. Either this file is not a zipfile, or it constitutes one disk of a multi-part archive. In the latter case the central directory and zipfile comment will be found on the last disk(s) of this archive. unzip: cannot find zipfile directory in one of 2020-08-28_17-26-14_GNSS-1.ubx.tag or 2020-08-28_17-26-14_GNSS-1.ubx.tag.zip, and cannot find 2020-08-28_17-26-14_GNSS-1.ubx.tag.ZIP, period. rm: cannot remove '': No such file or directory

Stefal, 2020-08-29:
Hi!
The Rinex conversion script need a zip file as input, and you try to convert a ubx file.
The actual conversion script is a quick and dirty one. Building a complete Rinex conversion tool is on my todo.

closed 2020-08-29 by Stefal

hcwinsemius, 2020-08-31:
Good to know. Should I somehow zip the ubx file then? I am not fully in the code base yet although I may dig into it. 

If you have an example how to get the rinex output from the front end I'd love to try it.

Stefal, 2020-08-31:
Every day, the ubx file from the previous day is zipped. Is it not the behaviour you see?