Stefal/rtkbase #108: self-recognition  USB communication device by vendor and product (/dev/tty...)

jancelin, 2020-11-20:
```/dev/ttyACM0``` is not a sure value, it is possible to fix a generic mounting point like ```/dev/ttyGNSS``` using the idVendor and the idProduct of the connected card.

On a raspbian (Debian) distribution, the file to modify is located here:

```/etc/udev/rules.d/99-com.rules```

for example by adding this configuration line to ```99-com.rules``` , the drotek DP0601 receiver connected in USB will always have a mount point equal to ```/dev/ttyGNSS``` : 

```SUBSYSTEM=="tty", ATTRS{idVendor}=="1546", ATTRS{idProduct}=="01a9", SYMLINK+="ttyGNSS"```

To know the idVendor and the idProduct of a material: 

```sudo cat /sys/kernel/debug/usb/devices | grep -E "^([TSPD]:.*|)$"```

example of use on building a flashable image for raspberry:

https://github.com/jancelin/pi-gen/commit/84c3a3607ee23b48563372505384865d3e1b36b7#diff-7a3cbbae1f6bffa1769577936cbd17d22be746f03df2c218e2c3acd872934598

https://github.com/jancelin/pi-gen/commit/84c3a3607ee23b48563372505384865d3e1b36b7#diff-277e41b735be4ef44428e3033a6a1061111abd5566e37df1312b9631eb1ec8c4

kikislater, 2020-11-23:
:+1: 
Works well. Since the rule is in rules.d folder you could create a new file instead of modifying existing one. It could help to deploy, avoid modifying original 99-com-rules and doesn't require the use of sed or other in custom deployment. 

Symlink could be shown with :
```
basegnss@basegnss:~ $ ls -lsa /dev/ttyGNSS 
0 lrwxrwxrwx 1 root root 7 Nov 23 16:00 /dev/ttyGNSS -> ttyACM0
```

Stefal, 2021-01-10:
What is the behaviour if you have 2 identical receiver connected?

closed 2024-01-02 by Stefal