Stefal/rtkbase #110: Failed to load resource: the server responded with a status of 400 (BAD REQUEST)

marcoridder14, 2020-12-16:
I can not save settings.
In the console lots of Failed to load resource: the server responded with a status of 400 (BAD REQUEST). URL http://192.168.1.124/socket.io/?EIO=3&transport=polling&t=NPjL-5X Message "The client is using an unsupported version of the Socket.IO or Engine.IO protocols"

Clean installation of Raspberry Pi OS Lite.
Chrome 86 87
Raspberry Pi 3B+ and Pi 4 tested, same problem.

1 month ago i did 2 installations and worked perfect.

Stefal, 2020-12-17:
Hi @marcoridder14,

I need to find some time to investigate. It seems related to https://github.com/miguelgrinberg/python-socketio/issues/578.

marcoridder14, 2020-12-17:
> Hi @marcoridder14,
> 
> I need to find some time to investigate. It seems related to [miguelgrinberg/python-socketio#578](https://github.com/miguelgrinberg/python-socketio/issues/578).

For your information, the image of jancelin works fine.

Stefal, 2020-12-31:
Should be fixed in the next release : https://github.com/Stefal/rtkbase/commit/f98766d825aebad0b4c950da33189226c3268961

closed 2020-12-31 by Stefal