Stefal/rtkbase #135: U-Blox_ZED-F9P_rtkbase.txt file is missing

vakhus, 2021-02-27:
**Describe the bug**
file
` rtkbase/receiver_cfg/U-Blox_ZED-F9P_rtkbase.txt` 
is missing

A setup like this:
[(https://www.ardusimple.com/connect-your-simplertk2b-to-a-raspberry-pi/)]
is not recognized.

`sudo ./install.sh -detect-usb-gnss --configure-gnss` 
gives
` Installation options: -detect-usb-gnss --configure-gnss ` 
` ################################`  
` CONFIGURE GNSS RECEIVER ` 
` ################################ ` 
` NO GNSS RECEIVER DETECTED, WE CAN'T CONFIGURE IT!`

Stefal, 2021-02-27:
Hi!
Your problem is not related to the txt file.
The F9P detection works only if it is connected with Usb and you're using an UART connection.
You should launch the configure script manually with your com port: 
`~/rtkbase/tools/set_zed-f9p.sh /dev/"your F9P_com_port" "your com speed" ~/rtkbase/receiver_cfg/U-Blox_ZED-F9P_rtkbase.cfg`
It could be something like:
`~/rtkbase/tools/set_zed-f9p.sh /dev/ttyS1 115200 ~/rtkbase/receiver_cfg/U-Blox_ZED-F9P_rtkbase.cfg`

vakhus, 2021-02-27:
Thank you for the quick reply.

I am trying to see which com port is used but I can not find it.
At the arduisimple tutorial 
[https://www.ardusimple.com/connect-your-simplertk2b-to-a-raspberry-pi/]
`/dev/serial0`
is used with the default baudrate 38400.
Shall I use the above?

Sry for the newbie questions
Thanks again

vakhus, 2021-02-27:
I get
`dmesg | grep ttyS
[    1.538499] fe215040.serial: ttyS0 at MMIO 0xfe215040 (irq = 37, base_baud = 62500000) is a 16550`

Stefal, 2021-02-27:
At first, you have to stop the str2str_tcp service if it is running:
`sudo systemctl stop str2str_tcp`
Then try this:
`~/rtkbase/tools/set_zed-f9p.sh /dev/serial0 38400 ~/rtkbase/receiver_cfg/U-Blox_ZED-F9P_rtkbase.cfg`
If it works, you could reboot the Raspberry, and on the settings webpage, enter `serial0` in the com port, and keep 115200:8:n:1 for the com port settings.

vakhus, 2021-02-27:
thank you very much!
I used ttyS0 and it seems to work.
The problem is that both files 
2021-02-27_17-58-40_GNSS-1.ubx
2021-02-27_17-58-40_GNSS-1.ubx.tag
are binary.
I think that only tag should had been binary?

Stefal, 2021-02-27:
Glad it works
.tag files are sometime useful for post processing with rtklib. You can ignore them if you don't need these tag-time files.

If everything is ok, you can close the issue.

vakhus, 2021-02-27:
This file
2021-02-27_17-58-40_GNSS-1.ubx
shouldn't be binary, right?
it is binary!

Stefal, 2021-02-27:
.ubx are ublox binary files
.tag are rtklib tag-time files

vakhus, 2021-02-27:
thank you,
Sorry but i am imbarrassed to ask one final question.
How can I get a non binary readable nmea file please?

vakhus, 2021-02-27:
in a previous experiment I could read ubx files like this:
'$GPVTG,182.42,T,,M,1.422,N,2.633,K,A*31
$GPGGA,140229.003454.32141,N,055248.31771,E,1,05,1.86,-1.4,M,32.4,M,,*7E
$GPGSA,A,3,28,30,13,05,07,,,,,,,,4.57,1.86,4.17*00
$GPGSV,3,1,10,05,52,252,24,07,31,059,37,08,10,047,,09,13,119,*7E
$GPGSV,3,2,10,13,43,333,30,14,70,129,29,15,10,307,,17,02,155,*79
$GPGSV,3,3,10,28,66,151,19,30,58,040,36*70
$GPGLL,3734.32141,N,02248.31771,E,140229.00,A,A*6C
$GPRMC,140230.00,A,3734.32117,N,06248.31766,E,1.334,182.49,281120,,,A*6E
$GPVTG,182.49,T,,M,1.373,N,2.542,K,A*3C'

vakhus, 2021-02-27:
Thanks again for your precious help.
Enjoy the rest of the weekend!

closed 2021-02-27 by vakhus