Stefal/rtkbase #136: F9P detected, but no data on the web panel

XPav, 2021-03-12:
I have a RPi4 with Raspian Buster running Rtkbase 2.2.0. It is plugged into a UBlox F9P, specifically the C099-F9P Application board.  Raspbian is fully updated.

When I plug the ublox into my Windows PC and run ublox control center on Windows, I can see satellites and everything is good.

When I plug it into the RPi and use rtkkbase, I get no sats on the status page.    No satellites, 0,0,0 position.

All other services are off besides "main".

How can I figure out what's going on here?

The ublox is detected: 
````
Installation options:  --detect-usb-gnss --configure-gnss
################################
GNSS RECEIVER DETECTION
################################
/dev/ttyUSB1  -  u-blox_AG_C099__ODIN-W2_DBTPSPIA
/dev/ttyACM0  -  u-blox_AG_-_www.u-blox.com_u-blox_GNSS_receiver
/dev/ttyUSB0  -  u-blox_AG_C099__ZED-F9P_DBTPSPHZ
################################
CONFIGURE GNSS RECEIVER
################################
GNSS RECEIVER DETECTED: /dev/ttyUSB0  -  u-blox_AG_C099__ZED-F9P_DBTPSPHZ
com_port='ttyUSB0'

````

Config looks good in the settings panel:
[image]

Here's the diagnostics:
```
 str2str_tcp.service : Active
● str2str_tcp.service - RTKBase Tcp
Loaded: loaded (/etc/systemd/system/str2str_tcp.service; enabled; vendor preset: enabled)
Active: active (running) since Fri 2021-03-12 08:26:04 MST; 1min 2s ago
Process: 444 ExecStart=/home/pi/rtkbase/run_cast.sh in_serial out_tcp (code=exited, status=0/SUCCESS)
Main PID: 486 (str2str)
Tasks: 2 (limit: 3860)
CGroup: /system.slice/str2str_tcp.service
└─486 /usr/local/bin/str2str -in serial://ttyUSB0:115200:8:n:1#ubx -out tcpsvr://:5015 -t 0 -fl /home/pi/rtkbase/logs/str2str_tcp.log

Mar 12 08:26:03 rtkbase systemd[1]: Starting RTKBase Tcp...
Mar 12 08:26:04 rtkbase run_cast.sh[444]: stream server start
Mar 12 08:26:04 rtkbase run_cast.sh[444]: 2021/03/12 15:26:22 [CW---] 0 B 0 bps (0) /dev/ttyUSB0 (1) waiting...
Mar 12 08:26:04 rtkbase systemd[1]: Started RTKBase Tcp.

-- Logs begin at Fri 2021-03-12 08:26:00 MST, end at Fri 2021-03-12 08:26:54 MST. --
Mar 12 08:26:03 rtkbase systemd[1]: Starting RTKBase Tcp...
Mar 12 08:26:04 rtkbase run_cast.sh[444]: stream server start
Mar 12 08:26:04 rtkbase run_cast.sh[444]: 2021/03/12 15:26:22 [CW---] 0 B 0 bps (0) /dev/ttyUSB0 (1) waiting...
Mar 12 08:26:04 rtkbase systemd[1]: Started RTKBase Tcp.

str2str_ntrip.service : Inactive
● str2str_ntrip.service - RTKBase Ntrip
Loaded: loaded (/etc/systemd/system/str2str_ntrip.service; disabled; vendor preset: enabled)
Active: inactive (dead)

-- Logs begin at Fri 2021-03-12 08:26:00 MST, end at Fri 2021-03-12 08:26:54 MST. --
-- No entries --

str2str_rtcm_svr.service : Inactive
● str2str_rtcm_svr.service - RTKBase rtcm server
Loaded: loaded (/etc/systemd/system/str2str_rtcm_svr.service; disabled; vendor preset: enabled)
Active: inactive (dead)

-- Logs begin at Fri 2021-03-12 08:26:00 MST, end at Fri 2021-03-12 08:26:54 MST. --
-- No entries --

str2str_rtcm_serial.service : Inactive
● str2str_rtcm_serial.service - RTKBase serial rtcm output
Loaded: loaded (/etc/systemd/system/str2str_rtcm_serial.service; disabled; vendor preset: enabled)
Active: inactive (dead)

-- Logs begin at Fri 2021-03-12 08:26:00 MST, end at Fri 2021-03-12 08:26:54 MST. --
-- No entries --

str2str_file.service : Inactive
● str2str_file.service - RTKBase File - Log data
Loaded: loaded (/etc/systemd/system/str2str_file.service; disabled; vendor preset: enabled)
Active: inactive (dead)

-- Logs begin at Fri 2021-03-12 08:26:00 MST, end at Fri 2021-03-12 08:26:54 MST. --
-- No entries --

rtkbase_archive.timer : Active
● rtkbase_archive.timer - Run rtkbase_archive.service everyday at 04H00
Loaded: loaded (/etc/systemd/system/rtkbase_archive.timer; enabled; vendor preset: enabled)
Active: active (waiting) since Fri 2021-03-12 08:26:03 MST; 1min 3s ago
Trigger: Sat 2021-03-13 04:00:00 MST; 19h left

Mar 12 08:26:03 rtkbase systemd[1]: Started Run rtkbase_archive.service everyday at 04H00.

-- Logs begin at Fri 2021-03-12 08:26:00 MST, end at Fri 2021-03-12 08:26:54 MST. --
Mar 12 08:26:03 rtkbase systemd[1]: Started Run rtkbase_archive.service everyday at 04H00.

rtkbase_archive.service : Inactive
● rtkbase_archive.service - RTKBase - Archiving and cleaning raw data
Loaded: loaded (/etc/systemd/system/rtkbase_archive.service; static; vendor preset: enabled)
Active: inactive (dead)

-- Logs begin at Fri 2021-03-12 08:26:00 MST, end at Fri 2021-03-12 08:26:54 MST. --
-- No entries --
```

XPav, 2021-03-12:
Update 1:  default baud rate for the C099's USB port is 460800 baud, not 115200.  Fixed that.

I can now connect via TCP through the rtkbase service via U-center and see all the satellites there, but the status page is still 0,0,0 and no satellites.

Stefal, 2021-03-12:
Hi!
Interesting! I think you're the first one to try RTKBase with a C099-F9P.

It looks like the script is a little lost with all these U-Blox entries. I think that the real F9P is on `/dev/ttyACM0`. You can try to enter `ttyACM0` inside com port, it could works if the com speed is right and if the F9P is already set to send rawx and sfrbx.
The other way is to try to configure it with the script (at first, stop the main service):
 `path_to_rtkbase/tools/set_zed-f9p.sh /dev/ttyACM0 115200 path_to_rtkbase/receiver_cfg/U-Blox_ZED-F9P_rtkbase.cfg`

Stefal, 2021-03-12:
Oh! I didn't see your second message before I sent the previous one.

Stefal, 2021-03-12:
If `/dev/ttyUSB0` is the correct port, and baud rate is 460800, then you can try:
`path_to_rtkbase/tools/set_zed-f9p.sh /dev/ttyUSB0 460800 path_to_rtkbase/receiver_cfg/U-Blox_ZED-F9P_rtkbase.cfg`

Or set the F9P manually with U-Center:
- disable nmea out
- disable sbas
- enable rawx
- enable sfrbx

bonus (not needed for rtkbase but useful for U-center)
- enable NAV_PVT
- enable NAV_SAT
- enable NAV_SIG

Then save the settings

XPav, 2021-03-12:
Running the config script with the correct baud rate made everything work.  I changed the baud rate on rtkbase to 115200, and then did a "Save Config" via U-Center for good measure.

I'm now checking to see everything comes back after a power cycle and I have also amended the name of the bug to point out that its a C099 with a different powerup baud rate.

Thanks for the help.

closed 2021-03-12 by XPav