Stefal/rtkbase #140: Free space cleanup deletes at wrong value

kamcnally, 2021-03-26:
**Describe the bug**
The script `archive_and_clean.sh` is intended to delete old archives if there is enough available space. The intended amount is specified in the comment as 500MB but the code actually specifies 5000MB. This results in old archives being deleted at the wrong time.

(in my case, running with less than 5GB free, the archives are always deleted) 

See (one too many 0's):
https://github.com/Stefal/rtkbase/blob/4bb5b2c675ea3a8c0db7b67e4d16f158c67ccd0b/archive_and_clean.sh#L14

**To Reproduce**
Steps to reproduce the behavior:
1. Ensure that free space > 500MB but < 5GB
1. Run until `archive_and_clean.sh` is run
1. See all files deleted

**Expected behavior**
Older files should remain.

**Software/Hardware (please complete the following information):**
 - OS: Raspberry Pi OS Lite - Buster
 - SBC: Raspberry Pi Zero W
 - GNSS Device: NEO-M8T
 - Connection type: USB

**Screenshots**
NA

**Additional context**
I suggest moving this value out to a setting in `settings.conf` so it can be configured.

Stefal, 2021-03-26:
Hi @kamcnally 
It's already fixed, see #132 

But you're right, this value could be in `settings.conf`

kamcnally, 2021-03-26:
Oh thanks. Not sure how I missed that issue...

Stefal, 2021-04-11:
Fixed with https://github.com/Stefal/rtkbase/pull/141

closed 2021-04-11 by Stefal