Stefal/rtkbase #143: Ntrip caster feature

gaslypierre, 2021-04-14:
I have just upgraded from simplertk2b wifi master, rtkbase so much better than it to manage and use i don't want to downgrade but lacks caster option, can it be also caster by itself like wifi master.
Thanks for this great software!

Stefal, 2021-04-14:
hi!
I'm glad that you found RTKBase useful.

If you enable the rtcm service, you can connect a rover directly to the base station. Perhaps you need to configure your SBC (raspberry or something else) as a wifi hotspot ; for this feature, there are a lot of tutorials, and/or some packages to do this with a Gui, like https://github.com/RaspAP/raspap-webgui
Could these solutions help you?

gaslypierre, 2021-04-14:
I'm using topcon AGI-4 couldn't get it connected with rtcm service, It just have ntrip configuration menu.
I will port forward to RTKBase like on wifi master, if you can add caster feature too...

Stefal, 2021-04-14:
Ok, I understand. I don't know how and when I could add this feature, but I'm adding it to my long todo list 🙂

gaslypierre, 2021-04-14:
Please do us a favor and take it with the ones with priorities :)
Thanks again and again...

gaslypierre, 2021-04-14:
This is the one has that feature:
https://github.com/nebkat/esp32-xbee

[image]

gaslypierre, 2021-04-14:
> Ok, I understand. I don't know how and when I could add this feature, but I'm adding it to my long todo list 🙂

Can i ask something more, i will delete it later don't know where to ask.
I obtained position from u-blox's survey-in which of the height i must use?
Elepsoid, mean sealevel, geoid sep?

[image]

Stefal, 2021-04-14:
Just for my curiosity: Why don't you use an already existing caster?

about your height value. We usually use the ellipsoid but it depends on the datum you want to use.

gaslypierre, 2021-04-14:
Thank you so much for help, I tried rtk2go but not getting stable connection.
Yes i can do terminal command line tests with pleasure but i'm not nearby tractor, tomorrow i can test...

Stefal, 2021-04-15:
Ok, to test this, and if your tractor isn't on the same networg (ie there is a modem inside the tractor), you have to:

- Create a rule inside your router to have a fixed ip address for RTKBase
- Create a rule inside your router to forward the port 2101 to RTKBase
- Write down your router external ip address

Then on the RTKBase command line, launch this command:
`str2str -in tcpcli://127.0.0.1:5015#ubx -out ntripc://test:pass@:2101/RTKMNT#rtcm3 -msg 1004,1005,1006,1008,1012,1019,1020,1033,1042,1045,1046,1077,1087,1097,1107,1127,1230 -p 47.0983869 -1.2655108 36.4 -opt -TADJ=1 -i RTKBase Ublox_ZED-F9P,2.2.0 -a ADVNULLANTENNA -t 0`
You have to change the values after the `-p`  (latitude longitude height) with your base coordinates. 

Keep your terminal opened or the command will stop.

Now, go inside your tractor, and set it to use 
- caster address : your router external ip
- port: 2101
- mount point: RTKMNT
- user: test
- password: pass

Let me know your test result!

gaslypierre, 2021-04-16:
[image]
[image]
On status windows it shows connecting, authentication then "Ntrip connected receiving data" normally after this stage i must get RTK but it's just staying on GPS status

Stefal, 2021-04-16:
The tractor doesn't give many informations. Could you test your Ntrip Caster with a tool like RTKnavi or another ntrip client, outside of your local LAN?

gaslypierre, 2021-04-16:
[image]
Lefebure, can you suggest any other app on android to test.

Stefal, 2021-04-16:
> Lefebure, can you suggest any other app on android to test.
More difficult to setup, but rtkgps+ can do the job, in single mode.

BTW : If you see this answer soon, replace the caster ip with 109.211.143.194
It's my own base, with the same command line.

gaslypierre, 2021-04-16:
Same with your IP too, ntrip connected, receiving data but nothing on rtk sidr

Stefal, 2021-04-16:
You won't get a RTK fix with my base, as it is too far away. But did your tractor received some messages?

Stefal, 2021-04-16:
This is what I get with Rtkgps+ and my caster:
[image]

gaslypierre, 2021-04-16:
I forgot to check sorry, yes it is getting data 4300 messages received

Stefal, 2021-04-16:
> I forgot to check sorry, yes it is getting data 4300 messages received
That's better, the caster is working.

If you didn't received some messages from your caster, perhaps there is something wrong with your lan configuration, port forwarding, firewall, something else.

gaslypierre, 2021-04-16:
I will check it thank you very much!!!

gaslypierre, 2021-04-28:
@Stefal i finally solve my network problem. But i have a different problem and don't know where to ask so using this page, on tractor i drive lines with 12 meter spacing using national rtk correction, then i switch to my base and go over tire track over the same first line again the second line is off by 10 centimeters, do you know why it can happen? Can it be because of my height set wrong?

Stefal, 2021-04-28:
> 
> 
> @Stefal i finally solve my network problem. 
So it works, that's a good news!

> But i have a different problem and don't know where to ask so using this page, on tractor i drive lines with 12 meter spacing using national rtk correction, then i switch to my base and go over tire [...]

I'm sorry, I can't help you for this problem. There are too many unknown factors. But I don't think it's a height problem.
You should start to check if you use the same datum between your base and the national rtk correction. If your base coordinate is using a wgs84 datum, and the national rtk correction a local datum, it could explain your offset. See this blog post I wrote some time ago, some informations are not relevant for you, but it could help you to understand: https://www.openstreetmap.org/user/StephaneP/diary/390290

Stefal, 2021-04-28:
btw: the next rtkbase release will include the local caster on the web interface.

gaslypierre, 2021-04-28:
Caster happened too quickly beyond my dreams, thank you very much for this great project!!

gaslypierre, 2021-04-28:
They are using ITRF96 datum and 2005.00 referance epoch, how can i match their settings, is it possible?

Stefal, 2021-07-10:
Added in https://github.com/Stefal/rtkbase/releases/tag/2.3.0

closed 2021-07-10 by Stefal