Stefal/rtkbase #146: Standalone RTK base

highfreq, 2021-04-30:
When RTKBase is used as standalone base would it be possible it sets authomatically base position getting it from current position?
I have to copy and paste position everytime. Maybe it is already possible but relly couldn't find a way.

thanks,

Corrado

Stefal, 2021-05-01:
Hi!
You change your base location regularly? 
RTKbase is designed for permanent and stationary base station. When a user install a new base, he post process a 24 hours log to get very accurate coordinates.

So, taking automatically the base position isn't possible if you want something accurate.

Is your need something like the "survey in" function available inside the F9P?

highfreq, 2021-05-02:
Hello Stefal, Great job!!!!!

We would like to be able to use it with a drone when we are out for work, if the absolute position is not great at the beginning it is not a problem, but i guess it would be getting better during the day and the relative position to the base antenna of the drone would be more accurate even at the beginning.
So an option, with a checkbox to authomatically get current antenna position would be very useful to all the people that want to use it not in a fixed position.
Let me know if you have any question.

regrds,

Corrado

highfreq, 2021-10-09:
Hello Stefal, does the last version of rtkbase introduce the autofill of antenna position by any chance?

smash0190, 2024-10-24:
Is this something that is supported now? Yes some survey in for a couple of minutes, then setting the position even if absolute position isn't that great would be very  useful because you would still get higher accuracy relative positions.