Stefal/rtkbase #155: Can take only single solution using multiple Rover receivers

ionkarolos, 2021-05-18:
Hello,

I have successfully installed the latest release of rtkbase software using a Raspberry Pi 4 and an Ardusimple simpleRTK2B ublox ZED-F9P dev board. The web interface works like a charm. I have enabled though the web interface the RTCM TCP Server stream using the default RTCM messages (1004,1005(10),1006,1008(10),1012,1019,1020,1033(10),1042,1045,1046,1077,1087,1097,1107,1127,1230), port and receiver options (-TADJ=1). Using an Emlid Reach RS2 receiver (also ZED-F9P embedded) i am trying to get a fixed solution. But unfortunately the only solution i get is Single not even DGPS or Float. The broadcasted corrections from rtkbase software seems to feed into the Emlid Receiver just fine, as the Emlid's software confirms to me (a green receiving corrections message with the correct base fixed coordinates that i have added to rtkbase web gui). Could you please provide me some more feedback? Any ideas about this strange issue? Thanking you in advance.

Stefal, 2021-05-18:
Hi!
The RS2 seems to accept tcp Rtcm3 signal, so it should work.
I know some Topcon, Trimble,... rover works with RTKBase too (via a Ntrip Caster).
So....i'm sorry, I have no clue.

I just can add that the next RTKBase release will add an integrated Ntrip Caster, that will be another way to test your setup.

ionkarolos, 2021-05-19:
Thanking you Stefal for your Answer. 

Could you please give me some advice about the RTCM Messages types that i must sent using str2str? Currently i am using the above messages: 1004,1005,1006,1007,1008,1012,1019,1020. What RTCM Messages are you sending to get a fix to your Topcon, Trimple, Leica GNSS Receivers? Thanking you again.

Stefal, 2021-05-25:
The default messages should work.
If you have a spare F9P, perhaps you should connect it to a computer, and try to use it as a rover with Rtklib and your base. If it doesn't work, there is something wrong with your base station.

closed 2024-01-31 by Stefal