Stefal/rtkbase #156: Hi.

richardklasens, 2021-05-25:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

richardklasens, 2021-05-25:
Hi. wonder if its possible to use an exsisting ntrip stream to get into rtkbase, convert it to MSM4 and send it out to the WWW. (i want to convert a running RTCM stream with MSM7 to MSM4 with less data/messages)

Stefal, 2021-05-25:
Hi!
It's possible with a rtcm stream and using the undocumented `in_ext_tcp option` of `run_cast.sh` (https://github.com/Stefal/rtkbase/blob/781e60de7a2d08f72b1d3061b583a362cdf5f4ed/run_cast.sh#L14), edit `str2str_tcp.service` (replacing `in_serial` with `in_ext_tcp`), and set the right value in `settings.conf` (ext_tcp_source and ext_tcp_port. Right now it's available only in the dev branch)
But with a Ntrip input, this is not easily possible. You would have to add an external ntrip feature in `run_cast.sh`, add the needed variables in `settings.conf` and edit the `str2str_tcp.service` to use this new feature.

richardklasens, 2021-05-25:
Well, that is a fast answer! thanks for that, 
Well it would be cool to have this in a feature update for later, for now i'll try to build str2str on my Ubuntu server and see if this will work, if it does, i can try to change all this as your idea to see if it will happen in rtkbase.

closed 2025-01-02 by richardklasens