Stefal/rtkbase #165: 'INSTALLING RTKBASE REQUIREMENTS' psutil install failiure on fresh PiOS

electroklas, 2021-07-12:
newbi, 
Tried to install 2.3.0 on a fresh PiOs (2021-05-07 RaspberryPi Desctop)
and got stuck on RTKBASE REQUIREMENTS installation process:

_Installing collected packages: psutil, Flask-WTF, Flask-SocketIO, Flask-Login, eventlet, Bootstrap-Flask
  Attempting uninstall: psutil
    Found existing installation: psutil 5.5.1
ERROR: Cannot uninstall 'psutil'. It is a distutils installed project and thus we cannot accurately determine which files belong to it which would lead to only a partial uninstall._

Any tips for a linux /phyton noob...

Stefal, 2021-07-13:
Thank you for this report.

Most users install the RpiOs lite which include less packages. Inside RpiOs desktop, there is an already existing old `psutil` release which can't be upgraded with `pip`.

As a workaround, you can try this `sudo apt remove python3-psutil` then restart the RTKBase installation.

electroklas, 2021-07-13:
Thanks, that solved my issues….

I should of course selected the lite version for this appoication, but i guess that i had a stroke of bad luck when selecting image.. ;)

commit ea99f217 (2021-07-25): Updating psutil requirement to fix #165
closed 2021-07-25 by Stefal