Stefal/rtkbase #166: run_cast.sh usedr/pass issue?

bradgass, 2021-07-20:
Running on an RPi3 upgraded from 2.2.0 here...

I've been running an NTRIP caster manually from the CLI, but decided to upgrade and give the UI config a try.  I did not upgrade to 2.2.90 first, missed that was even available unfortunately.

The upgrade seems to have worked, and took all the new NTRIP caster options correctly, but does not enable or start the str2str_local_ntrip_caster.service.  I did manually enable and start the service, but the UI did not reflect that it was running.  Toggling the NTRIP caster "on" appears to work, but when refreshing the page (or looking from another browser), it's back to showing that it's "off".

But, even when manually enabling and starting the caster service, the service status didn't show what I think the right command line options should have been - the configured user and password simply ran together.  Looking closer, I think a colon is missing on line 20 of run_cast.sh, between ${local_ntripc_user} and ${local_ntripc_pwd}.

In my install, line 20 of run_cast.sh now reads -
out_local_caster="ntripc://${local_ntripc_user}:${local_ntripc_pwd}@:${local_ntripc_port}/${local_ntripc_mnt_name}#rtcm3 -msg ${local_ntripc_msg} -p ${position}"

I did edit run_cast.sh to add the colon, and then the service started with what now looks like the right command line, but I haven't had time to test it yet.

Can anyone confirm/dispell my observation or help me diagnose why the UI configuration and service control doesn't seem to work as expected?  Was the upgrade to 2.2.90 needed prior to 2.3.0?

Thanks!

Stefal, 2021-07-24:
Hi!
Thank you for your report. Strangely, with test/test for user/password, it works.
I will fix this bug asap and release a bugfix release

closed 2021-07-24 by Stefal

Stefal, 2021-07-25:
Release 2.3.1 is available: https://github.com/Stefal/rtkbase/releases/tag/2.3.1
You can update from the settings web page

commit 007841ca (2021-07-24): Adding missing ":" between user and password

Should fix #166