Stefal/rtkbase #168: Web server port and IP should be configurable

satel-kalletuulos, 2021-08-02:
**Is your feature request related to a problem? Please describe.**
In the current (2021-08-02 version) web_app/server.py, web server port is hard coded to 0.0.0.0 with standard http port i.e. 80:
line 759: `socketio.run(app, host = "0.0.0.0", port = 80)`

**Describe the solution you'd like**
Both web server port and address should be configurable in settings.conf file.

**Describe alternatives you've considered**
The file settings.conf would have two additional entries, for example:
```
web_app_port=80
web_app_ip=0.0.0.0
```
**Additional context**

by, 2021-08-28:
Alternatively, please keep at least 80 free and use 8008 or any other alternative http port, thanks!

by, 2022-02-26:
Provided pull request: https://github.com/Stefal/rtkbase/pull/202

closed 2024-01-31 by Stefal