Stefal/rtkbase #173: Make gpsd a requirement

by, 2021-08-28:
Please make a standalone version of gpsd a requirement and don't provide parts of it in the package; this way, staying up-to-date with the gpsd repo version is much easier and does not show the usual API version discrepancies (incl. potential errors), thanks!

by, 2021-08-28:
Current example from cgps:
`cgps: WARNING gpsd release 3.22, API: 3.14, expected 3.23.1~dev`

by, 2021-09-03:
Did some further digging in the code and the actual challenge is that configuring for gpsd is bound to installing a gpsd package, even if you have already built gpsd from git all by yourself.
So my suggestion is to decouple (1) configuring and (2) installing gspd, thanks!

Stefal, 2021-09-09:
Hi!
So your problem is that `apt-get install gpsd -y` overwrite the gpsd release you compiled from the source?

by, 2021-09-09:
Correct. -- More precisely, it overwrites parts of it and other parts installed take precedence over the compiled and already installed (and more recent) version.

Stefal, 2021-09-09:
Ok.
I should find a way to check if there is an already installed gpsd release > v3.22 before installing it.

As RTKBase doesn't need an up to date gpsd release (it needs it only to configure the F9P and transmit date/time to the chrony daemon), it's a low priority issue.

Stefal, 2024-01-31:
Since Debian 11, RTKBase don't need gpsd from the testing branch. Closing

closed 2024-01-31 by Stefal