Stefal/rtkbase #174: Version de rtkbase

badaboom, 2021-09-04:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.
Dans centipede.fr ,la cartographie des bases,quand on clique sur une base, on a le numéro de la version de ublox mais pas celle de rtkbase.

badaboom, 2021-09-04:
Ça n'a pas suivi mon commentaire.. dans centipede.fr, la fiche de la base on a la version de ublox mais pas celle de rtkbase.

Exemple SGc , ublox 2.20 il manquerait rtkbase 2.31

Stefal, 2021-09-09:
Bonjour,

Si c'est la présentation sur centipede.fr qui nécessite des améliorations, alors je ne peux pas faire grand chose. C'est plutôt sur le dépot centipede qu'il faut créer un ticket : https://github.com/jancelin/centipede

closed 2021-09-09 by Stefal