Stefal/rtkbase #175: RTKbase img for raspberry Pi Zero - GPSD issue

samuelgirardin, 2021-09-06:
**Describe the bug**
gpsd failed on raspberry pi zero with img 2.2.0 or 2.3.1

**To Reproduce**
Steps to reproduce the behavior:
1. just install a new fresh version on raspi zero
2. ssh login
3. try `gpsd -V`
4. See error = **Illegal instruction**
5. try `cgps` or `gsmon`
6.  See error = **Segmentation fault**

**Expected behavior**
gpsd running, cgps or gpsmon working.

**Software/Hardware (please complete the following information):**
-PI zero 'wh' try on 2 differents boards - fresh  flash with 2.2.0 or 2.3.1 rtkbase imgs

**Additional context**
`apt-cache policy gpsd` print :
```
gpsd:
  Installed: 3.22-4~bpo10+1
  Candidate: 3.22-4~bpo10+1
  Version table:
 *** 3.22-4~bpo10+1 100
        100 http://httpredir.debian.org/debian buster-backports/main armhf Packages
        100 /var/lib/dpkg/status
     3.17-7+b1 500
        500 http://raspbian.raspberrypi.org/raspbian buster/main armhf Packages
```
My thought are 3.22-4 gpsd version installed is not compiled for raspi zero, with a fresh buster img `apt-get install gpsd` serve the 3.17 version.
I did a try yesterday to compile from source gpsd with the pi zero, I got the 3.22 version working fine with rtkbase, except I had to launch it manually with : `sudo  gpsd -b -n tcp://localhost:5015 -F /var/run/gpsd.sock ` . Indeed I'm really not an expert on linux...

samuelgirardin, 2021-09-07:
Finally work like a charm on my raspberry pi zero,  by first compiling GPSD from source and did a manual install (except the second part of step 6). cgps and gpsmon are ok now, no more segmentation fault.
[image]

Stefal, 2021-09-09:
Thanks for the information. I'll take a look when I will have some time.

Stefal, 2021-12-09:
I can reproduce the problem when using a clean Raspberry Pi os lite buster.

Stefal, 2021-12-19:
Ok, I can't fix that easily. The backport package seems compiled for armv7.

2m, 2023-06-09:
I did a clean install on a first generation Raspberry PI using the image from the releases page and everything worked great.

closed 2024-01-02 by Stefal