Stefal/rtkbase #176: Cant get RTKBase to work with UBLOX_NEO_M8T

Leijoma, 2021-09-09:
Hi,
I am trying to set up a basestation with RTKBase, a Raspberry Pi 3B+ and a U-Blox Neo M8T
I have almost got it to work, but the receiver position does ot show up in the status window. I get the satellite status, but Lat, Lon and Height is zero. 

In the settings tab i have set receiver type to: Ublox_NEO-M8T and receiver format to ubx.

With a serial tool i can see that the receiver is transmitting both NMEA and UBX messages and in U-center i can see position as well as other data when i stream the data over tcp from the Pi 

I have tried to configure the receiver manually and turned on all NMEA as well as UBX messages. 
I tried using the config tool to load the U-Blox_NEO-M8T_rtkbase.txt file with the config tool

The Open Streetmap layer is also empty, but the Arial_hybrid works fine.

I'm running out of ideas about what to do...

/Magnus

Stefal, 2021-09-09:
Hi!

As most base station are using a dual frequency GNSS receiver, the latest RTKBase releases switched to a more appropriate configuration to compute the base realtime location.

If you can't switch to a F9P and want to view this realtime location with your M8T, you can try to edit `rtkbase_ppp-static_default.conf` inside `rtkbase/web_app/rtklib_configs/`:

change `pos1-ionoopt` value from `dual-freq` to `brdc`

Then restart RTKBase with `sudo systemctl restart rtkbase_web.service`

BTW: nmea is unuseful for RTKBase but you have to enable RAW output. (I don't remember the exact settings for the M8T, perhaps RAWX and SFRBX)

Leijoma, 2021-09-09:
Thanks, i will test that. I do have one F9P ana one M8T. My gut feeling was to use the F9P in the Rover  and the M8T in the base station. Are there any benefits of doing the opposite? Best solution would of course to have  two F9P, but thats out of my budget at the moment...

Stefal, 2021-09-11:
I don't know what is the best solution with a M8T and a F9P.

BTW: I have some problems here too with the OpenStreetMap layer. It seems there is something wrong on the tiles provider server.

mhfernandes, 2022-02-28:
I'm trying to use RTKBASE without internet with M8T module to collect raw UBX data, I created an access point on the Raspberry Pi 3, I can see satellite bars, I would like to use an RTC battery to set the time or the GNSS itself, the problem is that the raw data is not saved, only when the Raspberry is connected to the internet, could you help me? is there a version of RTKLIB that could work standalone to save UBX files without internet connection or another way to direct files to DATA folder?

Thank you!
[image]

[image]

[image]

[image]

Sometimes the button to activate raw data recording turns yellow.

geofis, 2022-03-01:
Hello @mhfernandes 

As @Stefal has commented in other posts, the gpsd daemon is able to get the date/time from the GNSS receiver.

However, if the daemon is unable to listen on the same port where the Main Service broadcasts (TCP server), it won't update the date/time from the receiver.

I am not completely sure, but I think the gpsd daemon is hard coded to listen on port 5015, the default. If it is possible for you, try setting the TCP port of the main service to 5015, restart the gpsd service and see if the File service can be turned on.

Regards.

Stefal, 2022-03-01:
Hi @mhfernandes 
@geofis is right, please set the TCP port to 5015. Then read this: https://github.com/Stefal/rtkbase/issues/204#issuecomment-1055283415
If you have more question, ask them on the other issue, not this one.

mhfernandes, 2022-03-01:
Thanks for answering geofis, the date problem I can solve with a real time battery, my biggest problem is that despite RTKBASE showing satellite signal as shown in the photos, it is not recording the raw UBX data, it only works when the raspberry is connected to the internet.

mhfernandes, 2022-03-01:
Obrigado por responder @geofis, o problema da data que consigo resolver com uma bateria em tempo real, meu maior problema é que apesar do RTKBASE mostrar sinal de satélite conforme mostrado nas fotos, ele não está gravando os dados brutos do UBX, só funciona quando o framboesa está conectado para a internet.

mhfernandes, 2022-03-01:
Thanks for answering @geofis, the date problem I can solve with a real time battery, my biggest problem is that although RTKBASE shows satellite signal as shown in the photos, it is not recording the raw UBX data, it only works when the raspberry is connected to the internet.

mhfernandes, 2022-03-01:
I apologize, I'm new to github and I don't know how to proceed yet, I just made the comment because I found the post consistent with my doubts, I appreciate the answers!

mhfernandes, 2022-06-14:
708 / 5 000
Resultados da tradução
Hello, I'm also trying to use the Neo M8T with the RTKBASE system, I've tried several configurations without success, I would like to use the system to record raw data like rover and create a wi-fi access point to access information and settings, any suggestion?
I installed the latest version 2.3.4 of RTKBASE with the commands:

$ wget https://raw.githubusercontent.com/Stefal/rtkbase/master/tools/install.sh -O install.sh
$ chmod +x install.sh
$ sudo ./install.sh --all

I configured the M8T messages for UBX raws, but it seems to me that there is no communication between the GNSS card and the raspberry via UART.

If anyone can help I will be very grateful, because linux is not my strong point, I don't understand much.

[image]

[image]

Please help me!

mhfernandes, 2022-06-15:
Hello, I'm using the Neo M8T as a rover, I installed the RTKBASE system and everything is working perfectly, I would like to disable the files (.TAG) and leave only the files (UBX), I've tried some modifications in the system, but nothing still worked , can anybody help me?

Thanks!

[image]

[image]

Leijoma, 2022-06-15:
That is really interesting. Could you share the config of the M8T please?

Best regards

Magnus

mhfernandes, 2022-06-15:
Tipos de tradução
Tradução do texto
Texto de partida
1 238 / 5 000
Resultados da tradução
Hi Leijoma, sorry for the english translation.
1st- Download the ISO 2022-04-04-raspios-buster-armhf-lite.img
2nd-Install on an SD card and create the text document "SSH" without root extension to allow access via PUTTY
3-The Raspberry must be connected to the internet
4th-Access the link (https://github.com/Stefal/rtkbase) and install the RTKBase system with the commands:
$ wget https://raw.githubusercontent.com/Stefal/rtkbase/master/tools/install.sh -O install.sh
$ chmod +x install.sh
$ sudo ./install.sh --all
[image]

5th-Connect the M8T module to the Raspberry, in my case it was via UART
6º- Access the system with PUTTY and check which port is configured with the command (dmesg | grep tty) and insert this information in the RTKBase settings page.
7º- Access the browser and paste the link (https://docs.centipede.fr/docs/base/port_serie.html), there are instructions to enable the RaspberryPi serial ports and get the mount point.
[image]

8º-Configure (rtkbase/web_app/rtklib_configs/) as explained in "Stefal" at the beginning of this post.
9th-Create a Wi-Fi remote access point to access the system autonomously, the raspberry must generate a network and through a fixed IP you can access RTKBase.

That's basically it.

Hope this helps!

mhfernandes, 2022-06-15:
> Olá, estou usando o Neo M8T como rover, instalei a última versão do sistema RTKBASE (2.3.4) e está tudo funcionando perfeitamente, gostaria de desabilitar os arquivos (TAG) e deixar apenas os arquivos (UBX), Já tentei algumas modificações no sistema, mas nada ainda funcionou, não estou interessado em compactar ou converter os arquivos para RINEX, apenas salvar os dados (UBX) para pós processamento. Ajude-me?
> 
> Obrigado!
> 
> [image]
> 
> [image]

Problema resolvido, aditando uma linha no"run_cast.sh" não gera mais o "TAG", somente o UBX.

mhfernandes, 2022-11-02:
can anybody help me? I apologize for posting the question in the wrong topic, but I don't know how to make a new one on Github.
I'm using the RTKBase system with some changes to work with the M8T in Rover, I'm having a problem with the zoom of the cursor of the coordinates on the status screen, in previous versions RTKBase applied the zoom only on the GNSS cursor, now it shows the zoom with the target and cursor, how can i solve this, would someone show me a way?

[image]

closed 2024-01-31 by Stefal