Stefal/rtkbase #177: Satellite status not working, ardusimple F9P

sytem, 2021-09-16:
**Describe the bug**
Lost satellite status and map display on web gui. 

**To Reproduce**
Installed fresh image ( at first I was replacing corrupt sd-card, but happen also with another RPi + ardusimple that has worked before with older version) No data shown, but sending to rtk2go seems to be ok.

Have tried to reconfigure receiver with command ./tools/install.sh --detect-usb-gnss --configure-gnss

**Expected behavior**
See satellite status bar and map again

**Software/Hardware (please complete the following information):**
 - RTKBase release number 2.3.1
 - OS: from https://github.com/jancelin/pi-gen_RTKbase/releases/tag/BaseGNSS-RPi-2.3.1 
 - SBC: Raspberry Pi 3B+
 - GNSS Device: Ardusimple Ublox_ZED-F9P
 - Connection type: USB

Stefal, 2021-09-16:
I just did a fresh install on a Pi 3B+ with a resetted F9P. Everything is fine.

Which port is in use on your Pi ? Mine is ttyACM0.

btw: on the map, do you see the entire world, or nothing at all ?

sytem, 2021-09-16:
[image]

Completly empty interface. At the same time http://rtk2go.com:2101/SNIP::MOUNTPT?NAME=koivumaeki where this instance is sending, seems to be ok

ttyACM0 in use, as is was before corrupted SD and reinstall to new card.

Could there be some setting in F9P that is not resetting correctly with install script?

Diagnostic is showing lot of repeating message, I think I increased log level when I was looking if there is any data.

str2str_tcp.service : Active
● str2str_tcp.service - RTKBase Tcp
Loaded: loaded (/etc/systemd/system/str2str_tcp.service; enabled; vendor preset: enabled)
Active: active (running) since Wed 2021-09-15 10:18:03 CEST; 1 day 8h ago
Process: 525 ExecStart=/home/basegnss/rtkbase/run_cast.sh in_serial out_tcp (code=exited, status=0/SUCCESS)
Main PID: 557 (str2str)
Tasks: 2 (limit: 2063)
CGroup: /system.slice/str2str_tcp.service
└─557 /usr/local/bin/str2str -in serial://ttyACM0:115200:8:n:1#ubx -out tcpsvr://:5015 -t 3 -fl /home/basegnss/rtkbase/logs/str2str_tcp.log

Sep 16 19:08:52 basegnss run_cast.sh[525]: 2021/09/16 17:09:11 [CC---] 598553400 B 39480 bps (0) /dev/ttyACM0 (1) 3 clients
Sep 16 19:09:47 basegnss systemd-journald[143]: Suppressed 10 messages from str2str_tcp.service
Sep 16 19:09:47 basegnss run_cast.sh[525]: 2021/09/16 17:10:06 [CC---] 598834104 B 39226 bps (0) /dev/ttyACM0 (1) 3 clients
Sep 16 19:09:52 basegnss run_cast.sh[525]: 2021/09/16 17:10:11 [CC---] 598860316 B 39344 bps (0) /dev/ttyACM0 (1) 3 clients
Sep 16 19:10:47 basegnss systemd-journald[143]: Suppressed 10 messages from str2str_tcp.service
Sep 16 19:10:47 basegnss run_cast.sh[525]: 2021/09/16 17:11:06 [CC---] 599145108 B 39477 bps (0) /dev/ttyACM0 (1) 3 clients
Sep 16 19:10:52 basegnss run_cast.sh[525]: 2021/09/16 17:11:11 [CC---] 599171260 B 39162 bps (0) /dev/ttyACM0 (1) 3 clients
Sep 16 19:11:47 basegnss systemd-journald[143]: Suppressed 10 messages from str2str_tcp.service
Sep 16 19:11:47 basegnss run_cast.sh[525]: 2021/09/16 17:12:06 [CC---] 599456124 B 39336 bps (0) /dev/ttyACM0 (1) 3 clients
Sep 16 19:11:52 basegnss run_cast.sh[525]: 2021/09/16 17:12:11 [CC---] 599482244 B 39169 bps (0) /dev/ttyACM0 (1) 3 clients

-- Logs begin at Wed 2021-09-15 10:17:01 CEST, end at Thu 2021-09-16 19:11:52 CEST. --
Sep 15 10:18:03 basegnss systemd[1]: Starting RTKBase Tcp...
Sep 15 10:18:03 basegnss run_cast.sh[525]: stream server start
Sep 15 10:18:03 basegnss systemd[1]: Started RTKBase Tcp.
Sep 15 10:18:03 basegnss run_cast.sh[525]: 2021/09/15 08:18:22 [CC---] 0 B 0 bps (0) /dev/ttyACM0 (1) 127.0.0.1
Sep 15 10:44:26 basegnss systemd-journald[143]: Suppressed 11 messages from str2str_tcp.service
Sep 15 10:44:26 basegnss run_cast.sh[525]: 2021/09/15 08:44:45 [CC---] 264880 B 36826 bps (0) /dev/ttyACM0 (1) 4 clients
Sep 15 10:44:31 basegnss run_cast.sh[525]: 2021/09/15 08:44:50 [CC---] 287180 B 37965 bps (0) /dev/ttyACM0 (1) 4 clients
Sep 15 10:45:26 basegnss systemd-journald[143]: Suppressed 10 messages from str2str_tcp.service
Sep 15 10:45:26 basegnss run_cast.sh[525]: 2021/09/15 08:45:45 [CC---] 558856 B 40363 bps (0) /dev/ttyACM0 (1) 4 clients
Sep 15 10:45:31 basegnss run_cast.sh[525]: 2021/09/15 08:45:50 [CC---] 582840 B 40673 bps (0) /dev/ttyACM0 (1) 4 clients

thousands of similar lines removed...

Sep 16 19:10:37 basegnss systemd-journald[143]: Suppressed 10 messages from str2str_local_ntrip_caster.service
Sep 16 19:10:37 basegnss run_cast.sh[547]: 2021/09/16 17:10:56 [CW---] 599093396 B 39152 bps (0) 127.0.0.1 (1) waiting...
Sep 16 19:10:42 basegnss run_cast.sh[547]: 2021/09/16 17:11:01 [CW---] 599119500 B 39015 bps (0) 127.0.0.1 (1) waiting...
Sep 16 19:11:37 basegnss systemd-journald[143]: Suppressed 10 messages from str2str_local_ntrip_caster.service
Sep 16 19:11:37 basegnss run_cast.sh[547]: 2021/09/16 17:11:56 [CW---] 599404608 B 40618 bps (0) 127.0.0.1 (1) waiting...
Sep 16 19:11:42 basegnss run_cast.sh[547]: 2021/09/16 17:12:01 [CW---] 599430696 B 39184 bps (0) 127.0.0.1 (1) waiting...

str2str_rtcm_svr.service : Inactive
● str2str_rtcm_svr.service - RTKBase rtcm server
Loaded: loaded (/etc/systemd/system/str2str_rtcm_svr.service; disabled; vendor preset: enabled)
Active: inactive (dead)

-- Logs begin at Wed 2021-09-15 10:17:01 CEST, end at Thu 2021-09-16 19:11:52 CEST. --
-- No entries --

str2str_rtcm_serial.service : Inactive
● str2str_rtcm_serial.service - RTKBase serial rtcm output
Loaded: loaded (/etc/systemd/system/str2str_rtcm_serial.service; disabled; vendor preset: enabled)
Active: inactive (dead)

-- Logs begin at Wed 2021-09-15 10:17:01 CEST, end at Thu 2021-09-16 19:11:52 CEST. --
-- No entries --

str2str_file.service : Inactive
● str2str_file.service - RTKBase File - Log data
[... 242 lines cut ...]
Sep 16 19:09:47 basegnss python3[556]: RTKLIB 9b Attempting to start rtkrcv...
Sep 16 19:09:47 basegnss python3[556]: rtkrcv start successful
Sep 16 19:09:47 basegnss python3[556]: Starting coordinate and satellite broadcast
Sep 16 19:09:47 basegnss python3[556]: Trying to stop rtkrcv
Sep 16 19:09:47 basegnss python3[556]: RTKLIB 10a Attempting to stop rtkrcv...
Sep 16 19:09:47 basegnss python3[556]: rtkrcv stop successful
Sep 16 19:09:47 basegnss python3[556]: RTKLIB 10b Attempting to stop satellite broadcasting...
Sep 16 19:09:47 basegnss python3[556]: RTKLIB 10c Attempting rtkrcv shutdown
Sep 16 19:09:47 basegnss python3[556]: rtkrcv shutdown successful
Sep 16 19:09:47 basegnss python3[556]: Browser client disconnected
[... cut ...]
Sep 16 19:09:47 basegnss python3[556]: started : no
Sep 16 19:09:47 basegnss python3[556]: state : base
Sep 16 19:09:47 basegnss python3[556]: available_configs : []
Sep 16 19:09:47 basegnss python3[556]: system_time_correct : True
Sep 16 19:09:47 basegnss python3[556]: log_path : /home/basegnss/rtkbase/web_app/../data
Sep 16 19:09:47 basegnss python3[556]: Browser client disconnected
Sep 16 19:09:47 basegnss python3[556]: Browser client connected
Sep 16 19:09:47 basegnss python3[556]: RTKLIB 21 Trying to read previously saved state...

Stefal, 2021-09-16:
Hmmm everything seems fine in the logs. It looks like your web browser doesn't load the status page correctly, because you should see the map, even without any GNSS receiver connected.
Could you try with another web browser, without any ad/js/... blocker ?

sytem, 2021-09-17:
No difference between browsers: main firefox (yes, this has lot of blockers installed but nothing seems to be doing anything on these pages), Chrome and also Safari that is completly clean install. Same with another PC

Now when I looked web page with inspector, there are few JS errors:
[image]

But as those files are minized, there is nothing for me to understand.

Map div is completly empty.

sytem, 2021-12-18:
Now had time again to test this. with test instance, both local and vpn are working ok now (were also empty earlier, with no changes made. RPI was just unpowered for months) On production instance, I have only vpn access at the moment and there interface is empty.

Had to test production with local access next time I travell there

sytem, 2021-12-23:
No difference in production unit with vpn or local ip

closed 2022-02-22 by Stefal

biemond, 2022-05-04:
did we find out what it can be? , I have the same with ardusimple . all is fine except status.

also tried different browsers , operating system, ad blockers , in and outside iot network. nothing working.  
websockets gets data.

biemond, 2022-05-04:
[image]

Stefal, 2022-05-04:
@biemond 
Your problem seems different. Please open another issue.