Stefal/rtkbase #189: RTK base send uBlox binary messages to rtk2go instead of RTCM

Leijoma, 2022-01-16:
I have an Ublox_ZED-F9P connected to a Raspberry PI with RTKBase. 
I have configured the F9P with the config tool and the cfg file that comes with RTKBase.
Everything looks good in the web interface and RTKBase sends data to rtk2Go, but it sends the data in uBlox binare instead of RTCM.

This is what it look like at rtk2go:
STR;Enebyberg_01;Stockholm;uBlox;uBlox binary msgs;0;;SNIP;SWE;0.00;0.00;1;0;sNTRIP;none;N;N;0;

What am i doing wrong? Is there a config in RTKbase i have missed or is the receiver not configured correct?

Stefal, 2022-01-16:
Nothing wrong on my side: 
[image]

Stefal, 2022-02-10:
I just made a quick test with rtk2go and everything seems ok:

[image]

result on http://monitor.use-snip.com/?hostUrl=rtk2go.com&port=2101:

[image]

closed 2022-11-10 by Stefal