Stefal/rtkbase #192: No map background

cdkonecny, 2022-01-28:
**Describe the bug**
Everything seems to work except no background appears for map on web GUI.   Same whether maptiler key is used or not.
Zoom, pin, and crosshair are visible in grey field

**To Reproduce**
Steps to reproduce the behavior:
1. Go to http://192.168.*.**/status

**Expected behavior**
See street map 

**Software/Hardware (please complete the following information):**
 - RTKBase release number 2.3.1
 - Ubuntu 20.04.3 LTS
 - Raspberry Pi 4B 2GB
 - Ublox F9P]
 - Connection type: USB

**Screenshots**
[image]

**Additional context**
Have tried multiple machines/browsers.   Windows/ios/android

Stefal, 2022-01-29:
Yes, the tile provider have some problems since a few months. I hope @HurdmanBegins will fix this soon. The next RTKBase release will include another Osm tile.

But if you have entered a Maptiler key in settings.conf, it should work. This is Maptiler, this morning on my own base:
[image]

cdkonecny, 2022-01-29:
I have the same result/output with and without maptiler.   Even with only openstreetmap, it is just grey.   Am I missing something?

Stefal, 2022-01-29:
To use Maptiler, you should:
- get an api key from Maptiler
- enter this key in the file `settings.conf` : https://github.com/Stefal/rtkbase/blob/2f5e80987553ec3d3c05ea6adad5e29a2f99dd62/settings.conf.default#L22
- Select this layer (Aerial_Hybrid) on the map selector (top-right)

cdkonecny, 2022-01-29:
I have done that.   My problem is that I can’t even get a regular map to work much less maptiler.