Stefal/rtkbase #201: NMEA output

ktuulos, 2022-02-24:
**Is your feature request related to a problem? Please describe.**
I have some applications, which operate on NMEA feed. They don't understand binary messages. Currently RTKBase works only using binary messages and there is no ready-made option to enable also NMEA message forwarding. However, at least in F9P, it is possible to get both NMEA and binary messages, as I understand.

**Describe the solution you'd like**
There should be possibility to enable NMEA messages from GNSS receiver and those should be available in some TCP port.

**Describe alternatives you've considered**
Currently I have connected a separate GPS receiver to another Raspberry and serving NMEA from that.

**Additional context**
None.

Stefal, 2022-02-24:
Hi!
NMEA output on a base station ? What you want looks like a Rover, not a Base. Or I don't understand how you want to use RTKBase.

ktuulos, 2022-02-25:
NMEA is needed to certain applications, which e.g. track visible satellites by parsing NMEA feed. It is much easier to parse, than binary format.

Stefal, 2022-02-25:
Ok!
There are at least 2 ways to do this:

1.  Customize RTKBase:
- Edit https://github.com/Stefal/rtkbase/blob/master/web_app/rtklib_configs/rtkbase_ppp-static_default.conf to output nmea
- modify server.py to launch start_Base https://github.com/Stefal/rtkbase/blob/fa86a03661364d1e1fd79cd611f5f63816d61363/web_app/server.py#L499-L514
- modify server.py to disable the start_Base timeout https://github.com/Stefal/rtkbase/blob/fa86a03661364d1e1fd79cd611f5f63816d61363/web_app/server.py#L153-L158

3. Use existing tools as: 
- [gpsdecode](https://gpsd.gitlab.io/gpsd/gpsdecode.html) from gpsd. This tool can parse a RTCM stream

BTW: I'd like to have a sky view inside the status page. See #100

closed 2022-11-27 by Stefal