Stefal/rtkbase #214: Ntrip Caster service did not start

marcoridder, 2022-03-30:
**Describe the bug**
Ntrip Caster service did not start

**To Reproduce**
Steps to reproduce the behavior:
1. Update the base from 2.2 -> 2.3.0 -> 2.3.3
2. Setup the Ntrip Caster service options (username, password, mountpoint)
3. Click save
4. Enable the Ntrip Caster service service, button is green. After reload the button is off
5. Diagnostic: str2str_local_ntrip_caster.service: Inactive
6. File service can not be enabled, buttons switch directly to off
7. Rtcm service can not be disabled, i click on 'on' and now it can not switched to off

**Software/Hardware (please complete the following information):**
 - RTKBase release number 2.3.3
 - OS: Basegnss from jancelin
 - SBC Raspberry Pi 3B
 - GNSS Device: Ublox F9P
 - Connection type: usb

**Screenshots**
[image]
[image]
[image]
[image]
[image]

Stefal, 2022-04-01:
I'll try to repro on my side.

Stefal, 2022-04-01:
I can't reproduce here.
Did you clear your browser cache? If you don't do it, the javascript code running could be from the old release and could not work with the latest RTKBase release.

marcoridder, 2022-04-04:
It works thank you. Cache busting could be an idea for future versions?

closed 2022-04-04 by marcoridder