Stefal/rtkbase #223: OpenStreetMap not working on new installation

MyWebCityDK, 2022-04-17:
I did a new raspberry installation with a new blank SD card, installed RTKbase, updatede all packages etc....
OpenStreetMap is not showing any map when chosing. in website
Ortho HR (fr) is working fine.

Stefal, 2022-04-21:
Which RTKBase release are you using?
Did you clear the brower cache on the status page?

MyWebCityDK, 2022-04-21:
We use v2.3.4
After some hours the menu changed and it is now working with "OpenStreetMap (Osm)"
"OpenStreetMap (Hot)" is not working, it is just a blank map.
"Ortho HR (france)" is working fine.

Stefal, 2022-04-21:
For the HOT layer, it's a knows problem with the tile provider. It's the reason I've added the standard OpenStreetMap layer.

closed 2022-04-21 by Stefal