Stefal/rtkbase #225: Satellite status not working, ardusimple F9P

biemond, 2022-05-04:
Everything looks fine (settings + logs) only status page not. 

tried all browsers , clear cache , trust site , add basegnss.local  to dns etc., on same vlan and outside.

all service are up and journal looks ok. 

developer mode show websockets data  but page stay empty 
[image]

used https://github.com/jancelin/pi-gen_RTKbase/releases/tag/BaseGNSS-RPi-2.3.1 and then upgraded it.
got gps fix also from ucenter I see all sats and my lat and long.

so what can I try to fix this.

biemond, 2022-05-04:
[image]
[image]

Stefal, 2022-05-04:
- When you connect U-center to basegnss (port 5015), in the message view window, do you see the RAWX and SFRBX messages?

- On the status webpage, what do see in "coordinate message received" ?
Here is an example:
```
 Coordinate msg received: status.js:177:17
21:14:50,061 �[1mParameter:Value�[0m status.js:179:25
21:14:50,061 rtklib version:2.4.3 b34 status.js:179:25
21:14:50,061 rtk server thread:-1239047072 status.js:179:25
21:14:50,061 rtk server state:run status.js:179:25
21:14:50,061 processing cycle (ms):10 status.js:179:25
21:14:50,062 positioning mode:PPP-static status.js:179:25
21:14:50,062 frequencies:L1+L2 status.js:179:25
21:14:50,062 accumulated time to run:00:00:55.9 status.js:179:25
21:14:50,062 cpu time for a cycle (ms):1 status.js:179:25
21:14:50,062 missing obs data count:0 status.js:179:25
21:14:50,062 bytes in input buffer:0,0 status.js:179:25
21:14:50,062 # of input data rover:obs(56),nav(22),gnav(9),ion(30),sbs(0),pos(0),dgps(0),ssr(0),err(0) status.js:179:25
21:14:50,062 # of input data base:obs(0),nav(0),gnav(0),ion(0),sbs(0),pos(0),dgps(0),ssr(0),err(0) status.js:179:25
21:14:50,062 # of input data corr:obs(0),nav(0),gnav(0),ion(0),sbs(0),pos(0),dgps(0),ssr(0),err(0) status.js:179:25
21:14:50,062 # of rtcm messages rover: status.js:179:25
21:14:50,062 # of rtcm messages base: status.js:179:25
21:14:50,062 # of rtcm messages corr: status.js:179:25
21:14:50,063 solution status:PPP status.js:179:25
21:14:50,063 time of receiver clock rover:2022/05/04 19:15:08.999787163 status.js:179:25
21:14:50,063 time sys offset (ns):18587671884.801,0.000,36.120,0.000 status.js:179:25
21:14:50,063 solution interval (s):1.000 status.js:179:25
21:14:50,063 age of differential (s):0.000 status.js:179:25
21:14:50,063 ratio for ar validation:0.000 status.js:179:25
21:14:50,063 # of satellites rover:38 status.js:179:25
21:14:50,063 # of satellites base:0 status.js:179:25
21:14:50,063 # of valid satellites:10 status.js:179:25
21:14:50,063 GDOP/PDOP/HDOP/VDOP:2.7,2.2,1.2,1.9 status.js:179:25
21:14:50,063 # of real estimated states:236 status.js:179:25
21:14:50,063 # of all estimated states:236 status.js:179:25
21:14:50,064 pos xyz single (m) rover:4351652.231,-102588.153,4646380.177 status.js:179:25
21:14:50,064 pos llh single (deg,m) rover:47.06006410,-1.35047089,87.376 status.js:179:25
21:14:50,064 vel enu (m/s) rover:-0.015,-0.012,-0.023 status.js:179:25
21:14:50,064 pos xyz float (m) rover:4351652.231,-102588.153,4646380.177 status.js:179:25
21:14:50,064 pos xyz float std (m) rover:1.459,1.164,1.910 status.js:179:25
21:14:50,064 pos xyz fixed (m) rover:0.000,0.000,0.000 status.js:179:25
21:14:50,064 pos xyz fixed std (m) rover:0.000,0.000,0.000 status.js:179:25
21:14:50,064 pos xyz (m) base:0.000,0.000,0.000 status.js:179:25
21:14:50,064 pos llh (deg,m) base:0.00000000,0.00000000,0.000
```

biemond, 2022-05-04:
```
status.js:179 Parameter:Value
status.js:179 rtklib version:2.4.3 b34
status.js:179 rtk server thread:-1239182240
status.js:179 rtk server state:run
status.js:179 processing cycle (ms):10
status.js:179 positioning mode:PPP-static
status.js:179 frequencies:L1+L2
status.js:179 accumulated time to run:00:02:31.8
status.js:179 cpu time for a cycle (ms):1
status.js:179 missing obs data count:0
status.js:179 bytes in input buffer:0,0
status.js:179 # of input data rover:obs(0),nav(0),gnav(0),ion(0),sbs(0),pos(0),dgps(0),ssr(0),err(0)
status.js:179 # of input data base:obs(0),nav(0),gnav(0),ion(0),sbs(0),pos(0),dgps(0),ssr(0),err(0)
status.js:179 # of input data corr:obs(0),nav(0),gnav(0),ion(0),sbs(0),pos(0),dgps(0),ssr(0),err(0)
status.js:179 # of rtcm messages rover:
status.js:179 # of rtcm messages base:
status.js:179 # of rtcm messages corr:
status.js:179 solution status:-
status.js:179 time of receiver clock rover:-
status.js:179 time sys offset (ns):0.000,0.000,0.000,0.000
status.js:179 solution interval (s):0.000
status.js:179 age of differential (s):0.000
status.js:179 ratio for ar validation:0.000
status.js:179 # of satellites rover:0
status.js:179 # of satellites base:0
status.js:179 # of valid satellites:0
status.js:179 GDOP/PDOP/HDOP/VDOP:0.0,0.0,0.0,0.0
status.js:179 # of real estimated states:236
status.js:179 # of all estimated states:236
status.js:179 pos xyz single (m) rover:0.000,0.000,0.000
status.js:179 pos llh single (deg,m) rover:0.00000000,0.00000000,0.000
status.js:179 vel enu (m/s) rover:0.000,0.000,0.000
status.js:179 pos xyz float (m) rover:0.000,0.000,0.000
status.js:179 pos xyz float std (m) rover:0.000,0.000,0.000
status.js:179 pos xyz fixed (m) rover:0.000,0.000,0.000
status.js:179 pos xyz fixed std (m) rover:0.000,0.000,0.000
status.js:179 pos xyz (m) base:0.000,0.000,0.000
status.js:179 pos llh (deg,m) base:0.00000000,0.00000000,0.000
status.js:179 # of average single pos base:0
status.js:179 ant type rover:
status.js:179 ant delta rover:0.000 0.000 0.000
status.js:179 ant type base:
status.js:179 ant delta base:0.000 0.000 0.000
status.js:179 vel enu (m/s) base:0.000,0.000,0.000
status.js:179 baseline length float (m):0.000
status.js:179 baseline length fixed (m):0.000
status.js:179 monitor port:0
```

biemond, 2022-05-04:
I don't see it 
[image]

biemond, 2022-05-05:
I got RXM-RAWX but no RXM-SFRBX, also updated firmware
[image]

biemond, 2022-05-05:
somehow I see  RXM-SFRBX on ucenter 2 but not on u center and also not on my pi.

biemond, 2022-05-06:
I resolved my issue , looks like for my rtk, rxm was disabled by default,  RTK requires this (at least the status page), had to load the right config to the ardusimple simplertk2b , now it works
<img width="1230" alt="Screen Shot 2022-05-06 at 18 11 38" src="https://user-images.githubusercontent.com/1085004/167172396-814a1b32-24c8-4ad1-a04b-e4013bb87a
[image]
8c.png">
[image]

closed 2022-05-06 by biemond

Stefal, 2022-05-06:
Hi!
Your F9P was not configured correctly. RTKBase can configure it automatically... only if the F9P is connected via USB. It seems it' not your case. You could have try this command:
```rtkbase/tools/set_zed-f9p.sh /dev/ttyserial0 57600 rtkbase/receiver_cfg/U-Blox_ZED-F9P_rtkbase.cfg```

biemond, 2022-05-06:
great thanks