Stefal/rtkbase #237: Connection problem to the NTRIP server?!

EinZweistein, 2022-08-02:
I haven't used RTKBase for a while.
Now I wanted to use RTKBase again with the old settings. RASPIAN OS updates were probably made.
Well - the connection to the NTRIP-server (rtk2go.com) does not work.
The message in the diagnosis is:
 - Aug 02 13:51:49 raspberrypi python3[449]: Received service to switch {'name': 'ntrip', 'active': False}
 - Aug 02 13:51:49 raspberrypi python3[449]: Trying to stop service ntrip

The connection and the data stream to the NTRIP server work with another program.

**Software/Hardware (please complete the following information):**
 - RTKBase release number 2.3.4
 - OS: 
> - PRETTY_NAME="Raspbian GNU/Linux 11 (bullseye)"
> - NAME="Raspbian GNU/Linux"
> - VERSION_ID="11"
> - VERSION="11 (bullseye)"
> - VERSION_CODENAME=bullseye
> - ID=raspbian
> - ID_LIKE=debian
> - 64 Bit
 - SBC Raspberry Pi 3 Model B Rev 1.2
 - GNSS Device: Ublox F9P
 - Connection type: usb

GwnDaan, 2022-08-03:
[image]

> * Aug 02 13:51:49 raspberrypi python3[449]: Received service to switch {'name': 'ntrip', 'active': False}
> * Aug 02 13:51:49 raspberrypi python3[449]: Trying to stop service ntrip

This only says that you put the ntrip service button in the screenshot above to **"off"**.  Can you provide more of the diagnostics? Also make sure that the 2 buttons in the screenshot above is set to the green **"on"**.

EinZweistein, 2022-08-03:
Thanks. Yes, both set On green.
Tried restarting several times.
A data stream with the same settings (GNSS base receiver and NTRIP (rtk2go.com) works with the "SNIP" software.

[diagnostics.txt](https://github.com/Stefal/rtkbase/files/9251528/diagnostics.txt)

GwnDaan, 2022-08-03:
Hmm, I am unable to reproduce this issue. I don't see why it's trying to stop the ntrip service all the time from the diagnostics you sent. 

I think the fastest way for you to fix this, would be to reinstall RTKBase and copy over your old settings. I suspect updating to the latest version went wrong somehow. If reinstalling is not an option, @Stefal might know a fix for this.

EinZweistein, 2022-08-04:
Hi, Thank You. I also had the idea of reinstalling. After the third try, I opened this post. Only the fourth time it worked again... ???
Thanks

closed 2022-08-04 by EinZweistein