Stefal/rtkbase #238: Is it possible to add message 1013?

lansalot, 2022-08-04:
Evening all!

Just came across rtkbase the other day and amazed at the simplicity. Using an rpi an C099-F9P for testing, I had a base station up and running and casting to rtk2go.com pretty quickly. However, I tried an FJDynamics auto-steer (data from Soil Essentials) for one tractor, and a Trimble for another - and neither will consume the RTK signal. Both act like it can't be found and won't get a fix. From poking a hole in my firewall, I've been able to watch via netstat the incoming connections on the rpi via ssh to 2101, so I know they're getting there.

Soil Essentials puts out messages 1004, 1006, 1008, 1012, 1013 and 1033 and rtkbase puts out 1004, 1006, 1008, 1012 and 1033 - so I'm missing 1013 from that list.
Trimble puts out 1004, 1005, 1008, 1012, 1013, 1032, 1033 and 1230 - so I'm missing 1013 and 1032. in that list.

I've read up on what 1013 did and it seemed to be "these messages are sent at these rates", and the wording suggested it wasn't absolutely necessary.

I put the Soil Essentials info in to the Trimble, and it didn't work - so from that I can infer that 1032 is necessary for Trimble. SNIP shows a binary string it's unable to decode, so I'm assuming that's a watermark or somesuch? The Trimble info worked fine on the FJDynamics tho.

From looking at everything else in RTCM 3 Message Viewer in SNIP, everything else looks to be in order

**Is your feature request related to a problem? Please describe.**
Missing 1013 preventing tractors connecting and obtaining a fix. Can this be added?

**Describe the solution you'd like**
Add 1013 message in the hope that it works

**Describe alternatives you've considered**
Added the 1013 in RTCM messages, but no deal which I assumed would happen because it's just not supported?

Also, the 1033 ADVNULLANTENNA shows "RTKBase Ublox_ZED-F9P", but the string in RTKBase receiver type is just "Ublox_ZED-F9P". Although I can override that with a command line instance of str2str, and knock it out in its entirety, if I put something else in the GUI it shows up as for example "RTKBase TRIMBLE_NETR5", and I need to put an underscore in, whereas an actual Trimble would have a space. Could this be extended so spaces can be utilised in receiver name, with option to drop the "RTKBase " prefix?

Lastly, and I'm probably doing this wrong, but passing receiver options, I had hoped I could just override the -i on there and it would take my own rather than the ones it wanted to supply, but that seems to throw the executable on running. Maybe I'm just doing that wrong tho...

GwnDaan, 2022-08-05:
Welcome @lansalot!

This is interesting, supporting more brands as correction signal receiving clients is definitely a good thing in my opinion, but it seems like there is no quick fix for supporting the message types (1013 and 1032) you mentioned.

> Added the 1013 in RTCM messages, but no deal which I assumed would happen because it's just not supported?

We currently use RTKLIB for encoding the rtcm3 messages. I didn't find any support for encoding rtcm 1013 nor 1032 in their code: https://github.com/tomojitakasu/RTKLIB/blob/71db0ffa0d9735697c6adfd06fdf766d0e5ce807/src/rtcm3e.c#L2312-L2416. This is why adding these types to the list on the web ui has no effect. I also checked a few of the popular RTKLIB forks; [rtklibexplorer](https://github.com/rtklibexplorer/RTKLIB/blob/a04369e6d499804192420d641c86c0616d88c8fe/src/rtcm3e.c#L2607-L2726) and [emlid](https://github.com/emlid/RTKLIB/blob/81b5a6155242e3718298b0f5aeb931224809fb48/src/rtcm3e.c#L2409-L2517), but they don't seem to support them either. 

> SNIP shows a binary string it's unable to decode, so I'm assuming that's a watermark or somesuch?

From a message list online, I found the following about 1032: 'This message provides the ECEF location of the physical antenna used' ([source](https://www.use-snip.com/kb/knowledge-base/rtcm-3-message-list/)). Therefore it could well be that trimble depends on this message. For 1013 it includes: 'Not many NTRIP devices send this message'. It would be akward for FJDynamics  to depend on this message as not many devices send it apparently but nevertheless it is still useful to support.

If I have some spare time in the next few weeks, I can look for possible solutions. However, I do not have access to the brands you mentioned, so unfortunately I cannot reproduce/solve this problem on my own.

> Could this be extended so spaces can be utilised in receiver name, with option to drop the "RTKBase " prefix?

Yes, I will create a PR for this feature soon.

> I had hoped I could just override the -i on there

You could try modifying the following line in the `run_cast.sh` file in the root folder of this project to overwrite the -i argument:
https://github.com/Stefal/rtkbase/blob/8c75592873ed87a3ef66db5b46a17ab08a8eb0ed/run_cast.sh#L49

smirkingman, 2022-08-05:
From the message numbers, Soil Essentials and Trimble seem to rely on RTCMV2 (i.e. legacy) messages, which leads me to suspect that they need the 1013 to figure out which messages will be arriving, rather than figuring them out from the messages actually received. See the warning "_Sending legacy (1001-1004 and 1009-1012) and MSM messages in the same stream can cause problems for remote RTK users and is not recommended._" at [Novatel.com](https://docs.novatel.com/OEM7/Content/Logs/RTCMV3_Standard_Logs.htm). 

The ZED-F9P neither generates nor accepts 1013 (page 17 of ZED-F9P_IntegrationManual_UBX-18010802.pdf), and it's the same for 1032, so an out-of-the-box solution doesn't seem on the cards.

As an aside, for 1032, I've noticed that Trimble in general alway requires an antenna description, often from a prescribed list, for example at [CenterPoint RTX Post-Processing](https://trimblertx.com/Home.aspx); they **should** however accept 1005 / 1006 instead.

It sounds like your suppliers need to modernise the RTK infrastructure that they accept, but I also realise that they are unlikely to be sympathetic to people under-cutting their expensive solutions (demanding 1013 and 1032 is an easy way to force you to buy their receivers >;-) ).

Sorry that I can't be of more help. If it's any use, I'm running an identical setup on RTK2GO, you're welcome to connect to [chgevymg8a ](http://rtk2go.com:2101/SNIP::MOUNTPT?baseName=chgevymg8a)as a point of comparison.

P.S. RTKBase is indeed an excellent piece of engineering, amongst the best I've seen in 4 decades of programming.

lansalot, 2022-08-07:
Thanks for the input there folks - will keep trying and maybe I can find one or two other devices to test it with. Also going to drop back to using u-center for a moment and see if I can get anything out that way at all.. Tho I suspect not, as that's what I had initially.

pyrog, 2022-08-08:
@lansalot Could you set 1033 AntennaDescriptor to "ADVNULLANTENNA" at least for a test ?

Note: See [What is ADVNULLANTENNA](https://www.use-snip.com/kb/knowledge-base/what-is-advnullantenna/).

lansalot, 2022-08-13:
Morning all!

A few hours spent at my mates yesterday and... success ! F9P now working as a correction source for FJ Dynamics, and Trimble (not sure of model, I think FMX 750) ! The Trimble is steering the Case directly through the tractor, the FJD in the Claas is a wheel-turner from an Android tablet. A fair bit of experimentation to get there, but.. it's working !

The FJD works on these parameters:
[image]

The Case/Trimble works on these:
[image]

Would be really interested to see if these work on other machines if anyone has any? rtk2go.com is no password, and I've got a direct hole in the firewall for the caster at aaaa:aaaa@lans.no-ip.biz:2101 mount point BALLOCH.

This document led me to the right settings for FJD: https://support.swiftnav.com/support/solutions/articles/44001907775-rtcm-output

Basically, it's using "MSM5". The Trimble, I thought I had it working with MSM4 but it dropped and wouldn't reconnect. Eventually got it with a combination I just came up with on experimentation. My laptop ran out of battery and I couldn't test any further.

FJD in action: https://youtu.be/cLvjmOE0-rY
Case in action: https://youtu.be/D7sSRx7XJ1Y

Distance from my station to the tractor - approx 18 miles.

lansalot, 2022-08-24:
This is depressing... I set up a new mount point at my friend's place, and off the bat, rtkbase worked with its default messages . This is exactly what they meant at school with "you're too clever for your own good" - before I'd found rtkbase, I was sniffing the messages that the providers were giving to the Case/Trimble and FJD systems and so naturally that's what I tried to feed them. Then I found rtkbase and continued down that blind alley.

If I'd left them on default, I could have saved a lot of time... :(

closed 2022-08-24 by lansalot