Stefal/rtkbase #240: Ability to push to multiple casters

lansalot, 2022-08-25:
Morning all

So, I've been experimenting with different message sets and tractors, and I've it working on FJ Dynamics, Trimble, and now John Deere. The first two work out of the box with the suggested message set, but if the JD spots any MSM in there, it really doesn't like it.

Any chance of being able to add multiple profiles for pushing different message sets to the caster? This could also be used to push even the same message set to multiple casters to provide a bit of redundancy, as I've noticed rtk2go do like a restart now and then...

I can see how to do it script-side, but am at a bit of a loss with the python web-side...

Another thing that might be useful is the ability to connect a separate rtkbase instance to port 5015 on the box connected to your GPS so it could be used for development? Eg, two raspberry pi, one connected to f9p and acting as it should, the other RPI connecting to it for GPS info (I can do this via CMD line of course, just a thought for experimentation in the GUI)

Stefal, 2022-08-31:
Hi,
1) In the next release you will have 2 ntrip output available. See https://github.com/Stefal/rtkbase/pull/244/
2) It's already possible with a hidden feature. Set the `ext_tcp_source` and `ext_tcp_port` in `settings.conf`, and edit the str2str_tcp service ExecStart line and change `in_serial` to `in_ext_tcp`
3) The process which compute the blue marker position stop if you leave the status page for 15 minutes. It should not restart the process if you refresh the page.

lansalot, 2022-09-03:
Thanks Stefal, that's superb! This will help folk a lot!

closed 2022-09-03 by lansalot