Stefal/rtkbase #248: Documented a setup on RPI

lansalot, 2022-09-20:
Not a feature request, but just to draw your attention to my attempt to document how to set up rtkbase for non-technical folks.

https://github.com/lansalot/FreeRTK/blob/main/README.md

Hopefully it might be of some use to anyone struggling. Thanks again for great software!

Stefal, 2022-10-12:
Nice job ! I hope it will be useful.

I have some concerns about the CRS used for the base coordinates, ITRF2014 is not the best choice in Europe, but It would be way more technical, and perhaps too difficult to understand for non technical users.

lansalot, 2022-10-12:
Yeah, it just picked that auto. But it seems to be OK. Hopefully it'll help some folks out.

closed 2024-01-16 by lansalot

Stefal, 2024-01-16:
Hi @lansalot 
Feel free to add a link to your guide in this new [wiki page](https://github.com/Stefal/rtkbase/wiki/External-ressources)