Stefal/rtkbase #249: Cant get RTKBase to work with U-Blox ZED_F9R

rtagrimike, 2022-10-03:
Hi,

I am trying to set up an RTK base station with RTKBase, a Raspberry Pi 3B+ and a U-Blox ZED_F9R
I have almost got it to work, but the receiver position doesn't show up in the status window. I don't get the satellite status, and Lat, Lon, and Height are all zero.

In the settings tab, I have set receiver type to: Ublox_ZED-F9R and receiver format to ubx.

With a serial tool, I can see that the receiver is transmitting both NMEA and UBX messages and in U-center I can see the position as well as other data.

I have tried to configure the receiver manually and turned on all NMEA as well as UBX messages.
I have done a firmware update on the F9R.
I have tried all sorts of settings in the settings tab.

I'm running out of ideas about what to do...

Has anyone else experienced these issues?
Has anyone else used the F9R?

Any and all help or advice is welcome.
/Mike

rtagrimike, 2022-10-04:
I think I figured out my issue. As shown below ZED_F9R does not output RTCM messages, which I believe to be what RTKBase uses.

[image]

I switched over to a lower-end GNSS chip that does support RTCM,  and now I'm away laughing. Hope this helps someone else down the line.

Stefal, 2022-10-05:
Hi!

RTKBase can auto configure a F9P if it's connected with usb before you launch the RTKBase installation. I don't know if it works too with a F9R. If it doesn't you can still configure it manually with U-Center.
These are the settings needed (no rtcm) : https://github.com/Stefal/rtkbase/blob/master/receiver_cfg/U-Blox_ZED-F9P_config_info.txt

closed 2024-01-31 by Stefal