Stefal/rtkbase #250: Survey In mode

tkerby, 2022-10-20:
Would be great to support the survey in mode of the ZED-F9P for those looking to use this as a more temporary local correction source (eg on site with a drone). Having the ability to trigger that with a switch to one of the Pi pins and some sort of status LED would be really useful. 

I'm also wondering if the reported position deviates a lot from the position entered by the user whether it should gate the outputs to avoid sending bad data. Could cover situations where a unit has moved or damage to an antenna for instance

DavideLentini, 2024-03-06:
@tkerby 
I make an update here to explain how I solved , hoping that I can be of help to other people .
I understand that the idea of this repository is that of a software designed for a fixed station , and therefore I think it will be quite complicated that in the future there will be a possibility of a survey system addition in .
So I thought that very simply the survey in phase, must be done BEFORE starting the rtkbase services .
Since I also work with a raspberry + ubuntu server+ ublox f9p , there is no GUI or even a command line interface to configure the survey .
So I devised a small script in python that initially stops rtkbase services afterwards sends the configuration parameters to start the survey , and does a RESET of the F9P to start it .
It is possible to follow the progress of the survey either with a led (using GPIO of the raspberry) or from the terminal , simply by going to read the UBX-NAV-SVIN packets .
Once the survey is finished , the python script will start the rtkbase services , passing access to the f9p's serial .
At this point from web interface , you can see the quality of the satellites , and start the various ntrip services. Since the f9p has finished the survey and is in TIME mode , it will send RTCM messages by itself .

Although it goes against the main idea of this repo , I did not use the automatic gnss configuration , because it gave me problems .

lansalot, 2024-11-01:
+1 to this idea. I've been asked to provide a mobile-base and having rtkbase give this functionality would be ideal!

Maybe even an option to save a list of locations, so returning to the same area a year later, they could call up the previous "fixed" location. Obviously, a perfectly accurate location wouldn't be needed here, just good-enough that they can refer to when onsite

smash0190, 2024-11-11:
Do you have this script anywhere?

DavideLentini, 2024-11-12:
> Do you have this script anywhere?

Hey! yes i have the script , but it's quite unuseful, i'll explain why. 

For the software i use , i needed a serial connection to the F9P for start the survey or receving all the data from the F9P, 
but i didn't think about the TCP port that RtkBase offers . This TCP port is a direct bidirectional connection to the F9P , so in my software i just add the link to the tcp server , and bingo , i have a "real" tcp/serial connection to the f9p

DavideLentini, 2024-11-13:
@krisgry 
Great script. 
Since RTKBase already offers a tcp connection , theoretically you could use external tools to initiate the survey.
For example , very simply using the u-center software , you can connect directly to the F9p with the tcp port offered by rtkbase , in this way you can start a survey and do many other things without problems.
The fact of adding a survey interface directly from the web interface would not be bad though

krisgry, 2024-11-13:
@DavideLentini I avoid u-center whenever I can, so I prefer ssh-ing into my base and running the script :slightly_smiling_face: