Stefal/rtkbase #265: Command for starting main service through terminal

rtagrimike, 2022-11-22:
I have a Raspberry Pi set up as a base station to which I do not have physical access. Sometimes when it reboots "main service" doesn't start, leaving the station offline.

I have it set up on PiTunnel, and now have remote access to the terminal. I am wondering if there is a command I can use to start "main service" from the terminal. Are there any commands to see the statuses of the different services?

Thanks for your help in advance

hcwinsemius, 2022-11-22:
There are several services that must be running. The provisioning script
show the commands to (re)start those as below. You replace start by status
to check if a service experiences any issues.

$ sudo systemctl start rtkbase_web
$ sudo systemctl start str2str_tcp
$ sudo systemctl start gpsd
$ sudo systemctl start chrony
$ sudo systemctl start rtkbase_archive.timer

Stefal, 2022-11-22:
@hcwinsemius is right with these commands. 
The one you need to restart the main service is ` sudo systemctl start str2str_tcp`

But it's strange that this service doesn't start at boot. The next time it happens, could you copy/paste here the output of ` sudo systemctl status str2str_tcp`?

closed 2023-02-08 by Stefal