Stefal/rtkbase #276: Ability to add antenna height to 1006

Kaeden16, 2023-01-20:
**Is your feature request related to a problem? Please describe.**
Currently if you use message 1006 antenna height is 0.0000. Would be nice to be able to insert a value

**Describe the solution you'd like**
Maybe a field to enter the value

**Describe alternatives you've considered**
SNIP can do this but requires use of P.F.A.T which is expensive.

Stefal, 2023-01-21:
Hi!
I which case do you need this message ? I don't see it as really useful for a base station, but I don't know all the use cases.

At first, I should check if the `str2str` antenna height parameter is translated to the 1006 rtcm message.
I've read some mixed result on this subject: https://github.com/tomojitakasu/RTKLIB/issues/312

Kaeden16, 2023-01-21:
The point that my base station is on was surveyed in. To the bottom of the antenna, from my understanding the antenna height would be the difference in the point to the phase center of the antenna which makes vertical accuracy better.

Stefal, 2023-01-22:
Ok, I understand.

Before I try to add this option, could you try to add manually your antenna height value in the str2str command line, and check that it works correctly ?

I don't know when I could add this feature. In the meantime, you could add this antenna height to your base elevation value.

Kcoolz, 2023-03-07:
I've ended up just using 1006 with PFAT which is fine for now. Also is there a possibility to add L5 message support? Im pretty sure it has to be enabled on the GPS to function.