Stefal/rtkbase #280: Display Spectrum analyzer info from F9P

Stefal, 2023-01-24:
With the latest firmware, a F9P can provide some spectrum informations. here inside U-center, 
antenna disconnected
[image]

antenna connected
[image]

Values like PGA could inform the user if the antenna is connected or if the antenna cable is broken.

Stefal, 2023-01-24:
ubx-mon RF could be useful too
[image]

exentler, 2023-02-03:
would FW 1.32 on F9P be mandatory on 2.4 release or is the for 2.5?

Stefal, 2023-02-03:
> would FW 1.32 on F9P be mandatory on 2.4 release or is this for 2.5?

The latest firmware isn't mandatory. If I add these spectrum analyzer features in a future release, these informations just won't be available for the Gnss base stations with an old firmware, but the rest of the RTKBase features will work normally.