Stefal/rtkbase #307: 2.4.1 update causes gui access impossibe ??

jmpalmito, 2023-03-01:
Bonjour,
Je viens de mettre à jour avec la dernière version directement depuis l'interface web (basegnss.local)
Ma base fonctionne (toujours en vert sur le réseau centipede). Accès ssh OK
mais impossible d'accèder à l'interface utilisateur (via ip ou basegnss.local)

Am I alone in this case ? something went wrong during the firmware update, I will consider to flash the SD CARD again. 
I try a software reboot (sudo reboot) it does'nt solve the problem.

Merci

jmpalmito, 2023-03-01:
Au risque de dire une bétise : il me semble que c'était directement de 2.3.4->2.4.1

$ sudo systemctl status rtkbase_web
● rtkbase_web.service - RTKBase Web Server
   Loaded: loaded (/etc/systemd/system/rtkbase_web.service; enabled; vendor preset: enabled)
   Active: activating (auto-restart) (Result: exit-code) since Wed 2023-03-01 13:18:28 CET; 5s ago
  Process: 1836 ExecStart=/usr/bin/python3 /home/basegnss/rtkbase/web_app/server.py (code=exited, status=1/FAILURE)
 Main PID: 1836 (code=exited, status=1/FAILURE)

 $ sudo /usr/bin/python3 /home/basegnss/rtkbase/web_app/server.py
/usr/local/lib/python3.7/dist-packages/_distutils_hack/__init__.py:33: UserWarning: Setuptools is replacing distutils.
  warnings.warn("Setuptools is replacing distutils.")
Traceback (most recent call last):
  File "/home/basegnss/rtkbase/web_app/server.py", line 59, in <module>
    from flask_bootstrap import Bootstrap4

Et je ne vois pas de trace de Bootstrap4

Stefal, 2023-03-01:
La transition se fait en 2 fois : jusqu'en 2.4.0, et ensuite de 2.4.0 vers 2.4.1.

Bref, ça m'interesse vraiment d'en savoir plus sur le problème.
Qu'est-ce que donne `sudo journalctl -u rtkbase_web`

Autre option, que je prenne le controle de la base à distance. Dans ce cas, il faut me contacter, par exemple par Telegram, ou en passant par le formulaire de contact de https://rtkbase.eu/index.php/contact/

jmpalmito, 2023-03-01:
Le journal dit la même chose : server.ply plante car ne trouve pas Bootstrap4 et systeme relance

 $ sudo journalctl -u rtkbase_web
-- Logs begin at Wed 2023-03-01 10:42:01 CET, end at Wed 2023-03-01 14:18:45 CET. --
Mar 01 10:42:15 basegnss systemd[1]: Started RTKBase Web Server.
Mar 01 10:42:18 basegnss python3[439]: /usr/local/lib/python3.7/dist-packages/_distutils_hack/__init__.py:33: UserWarning: Setuptools is replacing distutils.
Mar 01 10:42:18 basegnss python3[439]:   warnings.warn("Setuptools is replacing distutils.")
Mar 01 10:42:18 basegnss python3[439]: Traceback (most recent call last):
Mar 01 10:42:18 basegnss python3[439]:   File "/home/basegnss/rtkbase/web_app/server.py", line 59, in <module>
Mar 01 10:42:18 basegnss python3[439]:     from flask_bootstrap import Bootstrap4
Mar 01 10:42:18 basegnss python3[439]: ImportError: cannot import name 'Bootstrap4' from 'flask_bootstrap' (/usr/local/lib/python3.7/dist-packages/flask_bootstrap/__init__.py)
Mar 01 10:42:18 basegnss systemd[1]: rtkbase_web.service: Main process exited, code=exited, status=1/FAILURE
Mar 01 10:42:18 basegnss systemd[1]: rtkbase_web.service: Failed with result 'exit-code'.
Mar 01 10:42:58 basegnss systemd[1]: rtkbase_web.service: Service RestartSec=30s expired, scheduling restart.
Mar 01 10:42:58 basegnss systemd[1]: rtkbase_web.service: Scheduled restart job, restart counter is at 1.
Mar 01 10:42:58 basegnss systemd[1]: Stopped RTKBase Web Server.
Mar 01 10:42:58 basegnss systemd[1]: Started RTKBase Web Server.
Mar 01 10:43:00 basegnss python3[681]: /usr/local/lib/python3.7/dist-packages/_distutils_hack/__init__.py:33: UserWarning: Setuptools is replacing distutils.
Mar 01 10:43:00 basegnss python3[681]:   warnings.warn("Setuptools is replacing distutils.")
Mar 01 10:43:00 basegnss python3[681]: Traceback (most recent call last):
Mar 01 10:43:00 basegnss python3[681]:   File "/home/basegnss/rtkbase/web_app/server.py", line 59, in <module>
Mar 01 10:43:00 basegnss python3[681]:     from flask_bootstrap import Bootstrap4
Mar 01 10:43:00 basegnss python3[681]: ImportError: cannot import name 'Bootstrap4' from 'flask_bootstrap' (/usr/local/lib/python3.7/dist-packages/flask_bootstrap/__init__.py)
Mar 01 10:43:01 basegnss systemd[1]: rtkbase_web.service: Main process exited, code=exited, status=1/FAILURE
Mar 01 10:43:01 basegnss systemd[1]: rtkbase_web.service: Failed with result 'exit-code'.
Mar 01 10:43:31 basegnss systemd[1]: rtkbase_web.service: Service RestartSec=30s expired, scheduling restart.
Mar 01 10:43:31 basegnss systemd[1]: rtkbase_web.service: Scheduled restart job, restart counter is at 2.
Mar 01 10:43:31 basegnss systemd[1]: Stopped RTKBase Web Server.
Mar 01 10:43:31 basegnss systemd[1]: Started RTKBase Web Server.
Mar 01 10:43:33 basegnss python3[732]: /usr/local/lib/python3.7/dist-packages/_distutils_hack/__init__.py:33: UserWarning: Setuptools is replacing distutils.
Mar 01 10:43:33 basegnss python3[732]:   warnings.warn("Setuptools is replacing distutils.")
Mar 01 10:43:33 basegnss python3[732]: Traceback (most recent call last):
Mar 01 10:43:33 basegnss python3[732]:   File "/home/basegnss/rtkbase/web_app/server.py", line 59, in <module>
Mar 01 10:43:33 basegnss python3[732]:     from flask_bootstrap import Bootstrap4
Mar 01 10:43:33 basegnss python3[732]: ImportError: cannot import name 'Bootstrap4' from 'flask_bootstrap' (/usr/local/lib/python3.7/dist-packages/flask_bootstrap/__init__.py)
Mar 01 10:43:33 basegnss systemd[1]: rtkbase_web.service: Main process exited, code=exited, status=1/FAILURE
Mar 01 10:43:33 basegnss systemd[1]: rtkbase_web.service: Failed with result 'exit-code'.
Mar 01 10:44:03 basegnss systemd[1]: rtkbase_web.service: Service RestartSec=30s expired, scheduling restart.
Mar 01 10:44:03 basegnss systemd[1]: rtkbase_web.service: Scheduled restart job, restart counter is at 3.
Mar 01 10:44:03 basegnss systemd[1]: Stopped RTKBase Web Server.
Mar 01 10:44:03 basegnss systemd[1]: Started RTKBase Web Server.
Mar 01 10:44:05 basegnss python3[736]: /usr/local/lib/python3.7/dist-packages/_distutils_hack/__init__.py:33: UserWarning: Setuptools is replacing distutils.
Mar 01 10:44:05 basegnss python3[736]:   warnings.warn("Setuptools is replacing distutils.")
Mar 01 10:44:05 basegnss python3[736]: Traceback (most recent call last):
Mar 01 10:44:05 basegnss python3[736]:   File "/home/basegnss/rtkbase/web_app/server.py", line 59, in <module>
Mar 01 10:44:05 basegnss python3[736]:     from flask_bootstrap import Bootstrap4
Mar 01 10:44:05 basegnss python3[736]: ImportError: cannot import name 'Bootstrap4' from 'flask_bootstrap' (/usr/local/lib/python3.7/dist-packages/flask_bootstrap/__init__.py)
Mar 01 10:44:06 basegnss systemd[1]: rtkbase_web.service: Main process exited, code=exited, status=1/FAILURE
Mar 01 10:44:06 basegnss systemd[1]: rtkbase_web.service: Failed with result 'exit-code'.
Mar 01 10:44:36 basegnss systemd[1]: rtkbase_web.service: Service RestartSec=30s expired, scheduling restart.
Mar 01 10:44:36 basegnss systemd[1]: rtkbase_web.service: Scheduled restart job, restart counter is at 4.
Mar 01 10:44:36 basegnss systemd[1]: Stopped RTKBase Web Server.
Mar 01 10:44:36 basegnss systemd[1]: Started RTKBase Web Server.
Mar 01 10:44:38 basegnss python3[739]: /usr/local/lib/python3.7/dist-packages/_distutils_hack/__init__.py:33: UserWarning: Setuptools is replacing distutils.
lines 1-51

Je te donne l'accès SSH

Stefal, 2023-03-01:
Ok, pendant la mise à jour, les dépendances n'ont pas pu s'installer, à cause d'une histoire de chemin non trouvé:

```
basegnss python3[383]: user for RTKBase is:  basegnss
Feb 24 10:38:05 basegnss python3[383]: ################################
Feb 24 10:38:05 basegnss python3[383]: INSTALLING RTKLIB
Feb 24 10:38:05 basegnss python3[383]: ################################
Feb 24 10:38:05 basegnss python3[383]: No binary available for  armv7l . We will build it from source
Feb 24 10:38:05 basegnss python3[383]: ################################
Feb 24 10:38:05 basegnss python3[383]: COMPILING RTKLIB
Feb 24 10:38:05 basegnss python3[383]: ################################
Feb 24 10:38:06 basegnss python3[383]: RTKLIB-b34g/
Feb 24 10:38:06 basegnss python3[383]: RTKLIB-b34g/.gitattributes
```

```
basegnss python3[383]: user for RTKBase is:  basegnss
Feb 24 10:45:06 basegnss python3[383]: ################################
Feb 24 10:45:06 basegnss python3[383]: INSTALLING RTKBASE REQUIREMENTS
Feb 24 10:45:06 basegnss python3[383]: ################################
Feb 24 10:45:09 basegnss python3[383]: Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple, https://www.piwheels.org/simple
Feb 24 10:45:12 basegnss python3[383]: Collecting pip
Feb 24 10:45:13 basegnss python3[383]:   Downloading pip-23.0.1-py3-none-any.whl (2.1 MB)
Feb 24 10:45:26 basegnss python3[383]: Collecting setuptools
Feb 24 10:45:26 basegnss python3[383]:   Downloading setuptools-67.4.0-py3-none-any.whl (1.1 MB)
Feb 24 10:45:29 basegnss python3[383]: Collecting wheel
Feb 24 10:45:29 basegnss python3[383]:   Downloading wheel-0.38.4-py3-none-any.whl (36 kB)
Feb 24 10:45:30 basegnss python3[383]: Installing collected packages: pip, setuptools, wheel
Feb 24 10:45:30 basegnss python3[383]:   Attempting uninstall: pip
Feb 24 10:45:30 basegnss python3[383]:     Found existing installation: pip 20.1.1
Feb 24 10:45:31 basegnss python3[383]:     Uninstalling pip-20.1.1:
Feb 24 10:45:31 basegnss python3[383]:       Successfully uninstalled pip-20.1.1
Feb 24 10:45:38 basegnss python3[383]:   Attempting uninstall: setuptools
Feb 24 10:45:38 basegnss python3[383]:     Found existing installation: setuptools 47.1.1
Feb 24 10:45:38 basegnss python3[383]:     Uninstalling setuptools-47.1.1:
Feb 24 10:45:39 basegnss python3[383]:       Successfully uninstalled setuptools-47.1.1
Feb 24 10:45:41 basegnss python3[383]:   Attempting uninstall: wheel
Feb 24 10:45:41 basegnss python3[383]:     Found existing installation: wheel 0.34.2
Feb 24 10:45:41 basegnss python3[383]:     Uninstalling wheel-0.34.2:
Feb 24 10:45:41 basegnss python3[383]:       Successfully uninstalled wheel-0.34.2
Feb 24 10:45:41 basegnss python3[383]: Successfully installed pip-23.0.1 setuptools-67.4.0 wheel-0.38.4
Feb 24 10:45:48 basegnss python3[383]: WARNING: Requirement 'rtkbase/tools/wheel/cryptography-38.0.0-cp37-cp37m-linux_armv7l.whl' looks like a filename, but the file does not exist
Feb 24 10:45:48 basegnss python3[383]: Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Feb 24 10:45:48 basegnss python3[383]: Processing /rtkbase/tools/wheel/cryptography-38.0.0-cp37-cp37m-linux_armv7l.whl
Feb 24 10:45:48 basegnss python3[383]: ERROR: Could not install packages due to an OSError: [Errno 2] No such file or directory: '/rtkbase/tools/wheel/cryptography-38.0.0-cp37-cp37m-linux_armv7l.whl'
Feb 24 10:45:50 basegnss python3[383]: ERROR: Could not open requirements file: [Errno 2] No such file or directory: 'rtkbase/web_app/requirements.txt'

```

A possible fix is to run this command:
```sudo ~/rtkbase/tools/install.sh --rtkbase-requirements```

I see some other errors after the restart:
```
Mar 01 15:00:21 basegnss systemd[1]: Stopped RTKBase Web Server.
Mar 01 15:00:22 basegnss systemd[1]: Started RTKBase Web Server.
Mar 01 15:00:23 basegnss python3[10095]: Traceback (most recent call last):
Mar 01 15:00:23 basegnss python3[10095]:   File "/home/basegnss/rtkbase/web_app/server.py", line 33, in <module>
Mar 01 15:00:23 basegnss python3[10095]:     import eventlet
Mar 01 15:00:23 basegnss python3[10095]:   File "/usr/local/lib/python3.7/dist-packages/eventlet/__init__.py", line 17, in <module>
Mar 01 15:00:23 basegnss python3[10095]:     from eventlet import convenience
Mar 01 15:00:23 basegnss python3[10095]:   File "/usr/local/lib/python3.7/dist-packages/eventlet/convenience.py", line 7, in <module>
Mar 01 15:00:23 basegnss python3[10095]:     from eventlet.green import socket
Mar 01 15:00:23 basegnss python3[10095]:   File "/usr/local/lib/python3.7/dist-packages/eventlet/green/socket.py", line 4, in <module>
Mar 01 15:00:23 basegnss python3[10095]:     __import__('eventlet.green._socket_nodns')
Mar 01 15:00:23 basegnss python3[10095]:   File "/usr/local/lib/python3.7/dist-packages/eventlet/green/_socket_nodns.py", line 11, in <module>
Mar 01 15:00:23 basegnss python3[10095]:     from eventlet import greenio
Mar 01 15:00:23 basegnss python3[10095]:   File "/usr/local/lib/python3.7/dist-packages/eventlet/greenio/__init__.py", line 3, in <module>
Mar 01 15:00:23 basegnss python3[10095]:     from eventlet.greenio.base import *  # noqa
Mar 01 15:00:23 basegnss python3[10095]:   File "/usr/local/lib/python3.7/dist-packages/eventlet/greenio/base.py", line 470, in <module>
Mar 01 15:00:23 basegnss python3[10095]:     from OpenSSL import SSL
[... cut ...]
Mar 01 15:00:23 basegnss systemd[1]: rtkbase_web.service: Main process exited, code=exited, status=1/FAILURE
Mar 01 15:00:23 basegnss systemd[1]: rtkbase_web.service: Failed with result 'exit-code'.
Mar 01 15:00:53 basegnss systemd[1]: rtkbase_web.service: Service RestartSec=30s expired, scheduling restart.
Mar 01 15:00:53 basegnss systemd[1]: rtkbase_web.service: Scheduled restart job, restart counter is at 180.
Mar 01 15:00:53 basegnss systemd[1]: Stopped RTKBase Web Server.
Mar 01 15:00:53 basegnss systemd[1]: Started RTKBase Web Server.
```

Stefal, 2023-03-02:
Ok, j'ai compris. Le première version de RTKBase installée sur ce Raspberry Pi était une version 2.02 (ou précédente), qui n'ajoutait pas la variable d'environnement `rtkbase_path`. De ce fait, les chemins utilisés pour copier RTKlib et lancer l'installation des dépendances ne fonctionnaient pas.

La commande `sudo ~/rtkbase/tools/install.sh --rtkbase-requirements` a bien corrigé le problème, et j'ai ajouté la variable d'environnement avec `sudo echo "rtkbase_path=/home/basegnss/rtkbase" >> /etc/environment`

Ok, I get it. The first version of RTKBase installed on this Raspberry Pi was version 2.02 (or earlier), which did not add the rtkbase_path environment variable. As a result, the paths used to copy RTKlib and start installing the dependencies did not work.

The command `sudo ~/rtkbase/tools/install.sh --rtkbase-requirements` fixed the problem, and I added the environment variable with `sudo echo "rtkbase_path=/home/basegnss/rtkbase" >> /etc/environment`

closed 2023-11-04 by Stefal