Stefal/rtkbase #314: IGN PPP compute failure

Stefal, 2023-04-08:
**Describe the bug**
A rinex file sent to the IGN PPP online service fails because there is more than 24H observables.
A workaround is to delete 2 or 3 of the last observables to stay below 24H

Stefal, 2023-04-10:
To use `-te` parameter with convbin, full date is needed : `-te 2023/04/10 07:00:00`
I tried `-span 1` to limit time span but it doesn't work.

nubotix, 2023-05-09:
I forgot to mention I'm using RTKBase v2.4.1

 CSRS-PPP SPARK  was just succesfully processed the truncated file. (It rejected my previous untruncated attempts)

Thanks for your really useful software!

Dafydd

Stefal, 2023-05-10:
I just wanted to warn you that a lot of PPP tool works only with L1+L2 data. The M8T is L1 only.

Stefal, 2023-11-12:
One way to check if the rinex timespan is too long is to check these two lines : 
TIME OF FIRST OBS
and
TIME OF LAST OBS

then use a bash command like cut/awk/split/csplit

Stefal, 2023-12-29:
I've set file overlap default value to 0s

closed 2023-12-29 by Stefal