Stefal/rtkbase #318: Compatible hardware for base station

ffries, 2023-05-04:
Dear all,

I just discovered your project and I would like to set-up a base station for my local community in rural area in France. It would be nice along my LoraWan station to provide precise GPS.

It does not sound clear to me whether ublox hardware returning RAW Output available via UBX-RXM-RAWX Message would be suitable for RTKLIB.

In particular I purchased Uputronics Raspberry pie hat:
the https://store.uputronics.com/files/Uputronics%20Raspberry%20Pi%20GPS%20RTC%20Board%20Datasheet.pdf

If the Uputronics Raspberry pie hat works it would make a very suitable hat for the Raspberry Pi 2/3/4.

I am also planning to use a Raspberry Pi 2B which is still available cheap new or used.

So it would make a suitable base station for 20€ (Raspberry Pi 2B) and Uputronics Raspberry pie hat (50€).

Do you think it will work with RTKlib? I guess yes, but please confirm.

Kind regards,
FFries

Stefal, 2023-05-04:
Hi!
 - Yes, you can use a raspberry Pi 2B, or an Orange Pi Zero
 - No, the Uputronics Hat can't be used. I don't think it can output the RAW stream, and it's L1 only, a things of the past for RTK.
 - LoraWan usually doesn't offer enough bandwidth for a RTK correction signal.

If you live in France, do you know the [CentipedeRTK Project](https://centipede.fr/) ? RTKBase is directly connected with this open network.

ffries, 2023-05-04:
Sorry for the confusion, I am planning to use my Lorawan casing (where I have PoE), nothing more. Yes I plan to join [CentipedeRTK Project](https://centipede.fr/). Therefore I don't need a roaming solution with a base and a rover. Only a simple base (I renamed the post).

The Uputronics Hat can output RAW streams as written here:
https://store.uputronics.com/files/Uputronics%20Raspberry%20Pi%20GPS%20RTC%20Board%20Datasheet.pdf

Although it is part of the past, it could still be used and is very cheap. Please correct me if I am wrong but a stationary station does not require a RTK ublox chip, only RAW streams.

Also I am interested in moving the current software to docker under Raspberry Pi 2/3/4. I read your post about Balena cloud solutions and it seems promissing. But I would like to try docker-compose to ease installation on Raspberry pi.

Kind regards,
Ffries

Stefal, 2023-05-04:
If you already have this Uputronics Hat, you can play with it and try Rtkbase. If you don't have it, don't waste your money.

ffries, 2023-05-04:
Dear Stefal, No I don't have it I just ordered it. What would be a competitive solution with a more recent chip in that price range of 50€? 

Kind regards, Ffries

Stefal, 2023-05-04:
With L1 only receivers (base and rover), it take a long time to get a RTK Fix, and you will lose this Fix very easily. Moving from L1 to L1/L2 receiver is like upgrading your internet connection from adsl to optical fiber.

ffries, 2023-05-04:
Thank you for clarification.
I found this information: https://gnss.store/blog/post/rtk-ppp-and-autonomous.html

My centripedeRTK base will be mounted on a pole so it does not move.

A modern GNSS RTK receiver with two channels cost only 150EUR to 200EUR and it can reach centimeter precision in a matter of seconds. It might not be interesting to use software RTK as it can take hours an I may never reach the same precision as it is single channel. Furthermore RTK is no longer actively maintained since 2017 ... All work is focusing on hardware.

However I will try as I ordered the hardware and I only need to buy a new GNSS if it goes wrong.

So I am closing the issue and will reopen it when my work is finished.

Kind regards,
FFries

closed 2023-05-04 by ffries