Stefal/rtkbase #322: Allow 1007 message

82Jussi, 2023-06-13:
At least Topcon AGI-4 needed the 1007 alongside 1008, to be able to get the stable RTK fix.

I would like to be able to write to rtcm messages 1007 like other messages and get that message sent by that.

A clear and concise description of any alternative solutions or features you've considered.

[image]

Stefal, 2023-06-13:
> I would like to add 1007 message to rtcm message list and it would be sent with correction data. Am I wrong that it is not sent today?

It is sent if you add it to the message list, like you did on your first screenshot.

82Jussi, 2023-06-13:
Then this request is not needed. Thank you. Sorry for confusion.

closed 2023-06-13 by Stefal